ControlSystemStudio/cs-studio #2062: use {"longString":true} for support

NickeZ, 2017-02-01:
How do I `use {"longString":true} for support`? I tried putting it in the PV name but it didn't work. I can't find any property on the widget called `longString`. I'm trying to write a string to a waveform of chars.

```
2017-02-01 16:13:04.342 WARNING [Thread 602] org.diirt.support.ca.JCAChannelHandler (put) - You are writing the String ARM to BYTE channel IFCFASTINT:FSM-MSGS: use {"longString":true} for support
```

kasemir, 2017-02-01:
Use a text entry field with "Format: String".

That will handle byte arrays as strings, and works for both BOY and display builder, PVManager and vtype.pv.

For BOY and PVManager, you can also use "MyPVName {"longString":true}" as the PV name,

but that only works for the PVManager, you need to type it exactly so, including the spaces,

and when you later copy that PV into for example a command line 'caget' it won't understand that longString modifier.

-Kay

NickeZ, 2017-02-02:
I'm using BOY and I can't find a `text entry field`. Do you mean `Text Input` widget? I can't use that since this has to be a button that sends the message. Different buttons sends different text messages to the same PV.

Your second solution works, so thank you!

closed 2017-02-02 by NickeZ