ControlSystemStudio/cs-studio #2069: Accessing CS-Studio

claudio-rosati, 2017-02-06:
Hello,

I need some help in understanding if CS-Studio has what it is needed for the following scenario.

We want a Linux machine starting with a sort of "guest" account and CS-Studio automatically starting showing some read-only OPIs.

A user (part of one of the following groups: "guest", "operator", "engineer", or "root") should be able to login into CS-Studio (i.e. giving its credentials to CS-Studio through a "login" button somewhere) and having these originally read-only OPIs partially/totally changing to read-write depending on the user's group. When the user has finished his job, logs out and CS-Studio reverts to the original read-only OPIs.

Is that something already available on CS-Studio? We are planning to use RBAC system backed by LDAP. Is it supported?

Thank you for your help,
Claudio

kasemir, 2017-02-06:
Everybody would agree that this is nice to have.
Technical issue: Tthe authorization for PVs is handled on the Channel Access level. The CA client libraries use the current host name and the actual user for authorization.
In CSS we already have an interface to the standard JAAS authentication, i.e. you can for example interface with LDAP,  and we allow several ways of handling authorization, for example based on a plain file or LDAP groups etc., see http://cs-studio.sourceforge.net/docbook/ch13.html.

We do use that for the alarm system,  but for CA there's no way to say "I am now acting as user Fred" when you are really user "guest".
The first step would be an update to the channel access (and pv access) client libraries to have an API that allows submitting a user/password, or some other way to change the effective user. At a previous EPICS meeting @ SLAC, those who were at that time in the ESS controls group proposed that they'd in fact add RBAC to EPICS base. Not sure how far this work progressed.

claudio-rosati, 2017-02-06:
Thank you @kasemir, I'll forward your reply to my stakeholder that raised the question.

veotax, 2018-05-09:
[jCasbin](https://github.com/casbin/jcasbin) is an authorization library that supports models like ACL, RBAC, ABAC.

Related to RBAC, casbin has several advantages:
1. roles can be cascaded, aka roles can have roles.
2. support resource roles, so users have their roles and resource have their roles too. role = group here.
3. the permission assignments (or policy in casbin's language) can be persisted in files or database.
4. multiple models like ACL, BLP, RBAC, ABAC, RESTful are supported.

And you can even customize your own access control model, for example, mix RBAC and ABAC together by using roles and attributes at the same time. It's very flexible.

I saw there's a RBAC need here, and I think jCasbin is a good choice. What do you think? Thanks.