ControlSystemStudio/cs-studio #2081: Many Diirt temporary files

willrogers, 2017-02-13:
After @claudio-rosati's (very useful) work on diirt, I'm getting a lot of temporary files left behind in `/tmp`.  Could we do better at cleaning up?

https://docs.oracle.com/javase/7/docs/api/java/nio/file/Files.html#createTempDirectory%28java.nio.file.Path,%20java.lang.String,%20java.nio.file.attribute.FileAttribute...%29

[image]

claudio-rosati, 2017-02-13:
[#2083](https://github.com/ControlSystemStudio/cs-studio/pull/2083) will close this issue.

Now there is the new method DIIRTPreferences.toFiles(File,boolean) where the last parameter tells wheter add a system hook to delete the given folder when the application exits.

@shroffk 
I've already updated DiirtStartup.preWindowOpen to call this method.

@kasemir 
Kay you should do the same where previously DIIRTPreferences.toFiles(File) was called.

willrogers, 2017-02-13:
Very quick!  Thanks.

claudio-rosati, 2017-02-13:
👍

kasemir, 2017-02-13:
The SNS app calls DIIRTPreferences.toFiles(File) from the same DiirtStartup.preWindowOpen(), nothing I need to change.

Created #2086 for the preference GUI inconsistency so we can track its progress there.

berryma4, 2017-02-13:
Can we also have this on 4.4.x?
Thank you!

claudio-rosati, 2017-02-13:
How?

berryma4, 2017-02-13:
Cherry pick.

claudio-rosati, 2017-02-13:
OK, I'll google about hot to

claudio-rosati, 2017-02-14:
@berryma4 
I don't understand how to do cherry pick with SourceTree, it doesn't work and, in any case, I cannot find the 4.4.x branch... I made something bad in trying performing cherry pick?

claudio-rosati, 2017-02-14:
@berryma4 
I've tried to use SourceTree to redo what I've done in console (following some hints found on the web about cherry pick) and I now realized I've not cleaned my forked version but **I've removed 4.4.x from the original repository**.
I'm terribly sorry about that. I hope you've a local copy to start over again.

berryma4, 2017-02-14:
I can do the cherry pick.  I made a new issue ticket for the deleted 4.4.x branch. #2088

claudio-rosati, 2017-02-14:
OK, thank you

closed 2017-02-21 by berryma4