ControlSystemStudio/cs-studio #2086: How to set diirt preferences from plugin_customization.ini

kasemir, 2017-02-13:
The `vtype.pv` has recently been updated (#2065) to use the new diirt preferences (#2020).
This again allows configuring all EPICS settings from a `plugin_customization.ini` file,
either built into the product plugin or provided via the `-pluginCustomization` command line.

There's also again a preference GUI for the Channel Access settings, but it does not show the correct values.

**Example 1**
```
# plugin_customization.ini
org.csstudio.diirt.util.core.preferences/diirt.home=/tmp/bogus
org.csstudio.diirt.util.core.preferences/diirt.ca.addr.list=my_ca_gateway
```

_Result:_
`DIIRTPreferences.get().getString(ChannelAccess.PREF_ADDR_LIST)` returns "my_ca_gateway", vtype.pv can connect.
Preference GUI complains about missing diirt home (understandably) and shows "localhost" (wrong).

[image]

_Expected:_
Preference GUI should show "my_ca_gateway".

**Example 2**
Trying the "_default_" variant of the preferences since that was mentioned in #2020:
```
# plugin_customization.ini
org.csstudio.diirt.util.core.preferences/diirt.default.initialized=true
org.csstudio.diirt.util.core.preferences/diirt.home=/tmp/bogus
org.csstudio.diirt.util.core.preferences/_default_.diirt.ca.addr.list=my_ca_gateway
```

_Result:_
`DIIRTPreferences.get().getString(ChannelAccess.PREF_ADDR_LIST)` returns "localhost", so the plugin_customization.ini value is not used at all.

**Example 3**
```
# plugin_customization.ini
# /tmp/ diirt exists,
# /tmp/diirt/datasources/ca/ca.xml contains addr_list="test"
org.csstudio.diirt.util.core.preferences/diirt.home=/tmp/diirt
org.csstudio.diirt.util.core.preferences/diirt.ca.addr.list=my_ca_gateway
```

_Result:_
`DIIRTPreferences.get().getString(ChannelAccess.PREF_ADDR_LIST)` returns "my_ca_gateway", vtype.pv can connect.
Preference GUI no longer complains about missing diirt home (good) yet still shows "localhost" (wrong).

[image]

_Expected:_
Preference GUI should show "my_ca_gateway".

claudio-rosati, 2017-02-13:
As soon as I've finished with meters/gauges in Display Builder I'll fix this issues.

Moreover, I'm thinking to add a new preference (default to true) to enabled the automatic deletion of the temporary files (see #2081): when developing/debugging can be useful having the temporary files. Anyone contrary?

kasemir, 2017-02-13:
I think it's OK to always delete the temporary files.
For debugging, you can view them (or create a copy) while the application is still running.

willrogers, 2017-02-17:
I think I have a similar problem to Kay's number 1.

When I set the preference `org.csstudio.diirt.util.core.preferences/diirt.ca.server.port=6064` it seems to work (it connects to the right port) but it doesn't show that value in the preference page.

claudio-rosati, 2017-02-17:
I'm trying to solve some problems that are blocking me to publish a preliminary version of Meters and Gauges, then I'll look at this problem.

Hopfully it will be solved next week.

claudio-rosati, 2017-03-02:
@kasemir @berryma4 @willrogers 
I've started working on this issue starting from a strange behavior a colleague as seen on Windows (a platform ESS will not officially support):

> When the Data Sources category is selected inside the Properties dialog, an error dialog is opened telling that "The currently displayed page contains invalid vales".

 What really happens is that a NoClassDefFoundError is raised:

```
java.lang.NoClassDefFoundError: Could not initialize class org.csstudio.diirt.util.core.preferences.DIIRTPreferences
	at org.csstudio.diirt.util.preferences.DIIRTPreferencesPlugin.getPreferenceStore(DIIRTPreferencesPlugin.java:166)
	at org.csstudio.diirt.util.preferences.BasePreferencePage.doGetPreferenceStore(BasePreferencePage.java:214)
	at org.eclipse.jface.preference.PreferencePage.getPreferenceStore(PreferencePage.java:390)
	at org.csstudio.diirt.util.preferences.DataSourcesPreferencePage.createContents(DataSourcesPreferencePage.java:107)
	at org.eclipse.jface.preference.PreferencePage.createControl(PreferencePage.java:241)
	...
```

and this is something I don't really understand: DIIRTPreferencesPlugin.java:166 is just calling `DIIRTPreferences.get()`. At compile time there is no problem and the DIIRTPreferences class is correctly found. At runtime this seems being a problem on on Windows and not on MacOS X or Linux. 

It seems like the org.csstudio.diirt.util.core.preferences plugin for some reasons was not loaded on Windows.

Have you an idea on how I should proceed to understand/solve the problem?

berryma4, 2017-03-02:
Does the plugin `org.csstudio.diirt.util.core.preferences1.0.1.[date].jar` exist in the installed cs-studio/plugins directory?

kasemir, 2017-03-02:
Is there any "Caused by: .." reason for the exception? Could be that DIIRTPreferences is known, but it can't be initialized due to a missing dependency like org.apache.commons.lang3.StringUtils.

Can you run the product with the "-console" command line option and then use the command that dumps all the bundle states to see if org.csstudio.diirt.util.core.preferences plugin, commons.lang3, are all active? (forgot the name of the command. Type 'help' to get a list.)

claudio-rosati, 2017-03-03:
@berryma4 
Yes

@kasemir 
No, that is the problem. No other message on the log, but to the "-console" suggestion I discovered a problem (on my mac) that could be the cause of the windows behavior. I'll keep you informed.

Thank you

kasemir, 2017-03-03:
> .. "-console" .. discovered a problem ..
Alright, progress. You can also try "-consoleLog -console" together. That will get you a console, and in addition send all the log messages to that console.

On Windows there may be a problem getting the actual console, because normal windows programs don't have one. For the Eclipse IDE on Windows, you'll find both an 'eclipse.exe' and an 'eclipsec.exe' copy of the equinox launcher. The latter one supports the console (Windows cmd.exe type window). You can simply copy that eclipsec.exe as 'css.exe' into the windows product, and now you can run CSS on windows with a console.
Or use the telnet console support: Start CSS with '-consoleLog -console 9876' and now you can use 'telnet 9876' to connect to the console.

claudio-rosati, 2017-03-03:
Thank you.
I was able to track the problem on Windows with just the '-console', and I fixed it. This fix will be part of the pull request closing this issue.
Another good new is that I was able to track the problem expained by Example 1-3. A fix in on the way... probably Monday will it be available as pull request.

berryma4, 2017-03-14:
I'll cherry pick #2143 to 4.4.x

mvitorovic, 2017-04-25:
I just tested this again, and I still think a small tweak is needed:
if DIIRT doesn't contain the datasources/datasources.xml file, the code takes plugin_customization.ini values but complains about the the missing file on every CS-Studio startup.

But the moment the file is present, you only have two options:
if the file "empty":
<dataSources version="1"></dataSources>

the compositeDataSource defaults to "None" (source code), no matter what the plugin_customization.ini says.

If the XML contains the compositeDataSource tag, then plugin_customization.ini is ignored.

What I would like is some combination, where it does not complain about the missing file, but takes the value from the plugin_customization.ini (at least when creating the workspace, as is the case with the ca/ca.xml file). Afterwards it can of course use the save preferences.

claudio-rosati, 2017-04-25:
I'll investigate this. Give me some time because I'm a bit busy this and the following week.

mvitorovic, 2017-05-12:
Hello Claudio. Have you had the time to look at this yet? Thanks, Miha.

claudio-rosati, 2017-05-12:
Hello Miha,
I'm afraid but I had no time yet for looking at this problem. Next week there is IPAC '17 at Copenhagen and was busy on preparing my article.

claudio-rosati, 2017-05-12:
Miha,

can you give me a more detailed scenario to reproduce the problem?

My product (ESS CS-Studio) contains a good diirt directory embedded in the product itself, so everything is working.

What is exactly your situation? 
What is the initial content of your `plugin_customization.ini` file?
Have you a DIIRT folder pointed by the `plugin_customization.ini` file? What this foldera contains?

Claudio

mvitorovic, 2017-05-17:
Hello, sorry for not replying earlier - I was tied up in release issues.

My plugin_customization.ini contains:
== SNIP ==
org.csstudio.utility.pv/default_type=ca
org.csstudio.diirt.util.core.preferences/diirt.datasource.default=ca
org.csstudio.diirt.util.core.preferences/diirt.home=/opt/codac/css/css/configuration/diirt
org.csstudio.diirt.util.core.preferences/diirt.ca.addr.list=
org.csstudio.diirt.util.core.preferences/diirt.ca.beacon.period=15.0
org.csstudio.diirt.util.core.preferences/diirt.ca.max.array.size=460000
# defaults
org.csstudio.diirt.util.core.preferences/_default_.diirt.ca.auto.addr.list=true
org.csstudio.diirt.util.core.preferences/_default_.diirt.ca.addr.list=
org.csstudio.diirt.util.core.preferences/_default_.diirt.ca.pure.java=true
org.csstudio.diirt.util.core.preferences/_default_.diirt.ca.connection.timeout=30.0
org.csstudio.diirt.util.core.preferences/_default_.diirt.ca.beacon.period=15.0
org.csstudio.diirt.util.core.preferences/_default_.diirt.ca.repeater.port=5065
org.csstudio.diirt.util.core.preferences/_default_.diirt.ca.server.port=5064
org.csstudio.diirt.util.core.preferences/_default_.diirt.ca.max.array.size=460000
== SNIP ==

My /opt/codac/css/css/configuration/diirt contains the following items:
datasources/                                               <DIR>
datasources/datasources.xml                    <FILE>
datasources/beast                                      <DIR>
datasources/beast/beast.xml                      <FILE>

The two files contain:
datasources.xml (minus the comments):
== SNIP ==
<?xml version='1.0' encoding='UTF-8'?>
<dataSources version="1">
     <compositeDataSource defaultDataSource="ca" />
</dataSources>
== SNIP ==

beast.xml (minus the comments):
== SNIP ==
<?xml version="1.0" encoding="UTF-8"?>
<beast version="1">
     <dataSourceOptions configName="CODAC_AlarmHandler" />
</beast>
== SNIP ==

Our desire is to get rid of the datasources.xml files and have everything dictated from the plugin_customization.ini. (I guess the beast.xml has nothing to do with your part. If this is true, just ignore it for now)

claudio-rosati, 2017-05-22:
@mvitorovic I would like to comment your original post, but before I like to summarize how preferences works, just to be sure we are we are speaking about the same thing.

When the application starts, then:

- the "wired" configuration is read (from the plug-in's `preferences.ini` file);
- then `plugin_customization.ini` is read, possibly overriding some preferences;
- then the `org.csstudio.diirt.util.core.preferences.prefs` file from the workspace is read and applied (that file is created when the workspace is created and updated each time a change is made through the Preferences dialog).

Now I've done the following tests (inside `plugin_customization.ini` I've specified a DIIRT folder and I played with `datasources.xml`, while `org.csstudio.diirt.util.core.preferences/diirt.datasource.default` was set to `pva` to easily identify it):

- if `datasources.xml` is missing, then **_Default Data Source_** will be _PV Access_, as expected;
- if `datasources.xml` is empty (I mean an empty file), **_Default Data Source_** will be _PV Access_, as expected;
- if `datasources.xml` is an empty data source (I mean an xml file with the `dataSources` element but nothing inside it), **_Default Data Source_** will be _None_, as you reported.

In the first two cases JAXB will fail to read `datasources.xml` and nothing more happens. In the last case an instance of the DataSources object is created with its default values, and this is the problem.

I'll change the initialization code of these object to use the default values coming from `plugin_customization.ini`. I'll let you know when ready for test.

mvitorovic, 2017-05-23:
Thanks. Also for the missing missing files, maybe you could convert the stack trace into a warning, not to scare the users?

claudio-rosati, 2017-05-23:
Sorry I don't understand of which stack trace you're speaking about. If `datasources.xml` is missing I already havd 2 warnings:

```
2017-05-23 09:13:24.898 CONFIG [Thread 1] org.csstudio.diirt.util.core.preferences.DIIRTPreferences (resolveAndVerifyDIIRTPath) - DIIRT home path resolved [before: platform:/config/diirt, after: /Users/claudiorosati/Desktop/ESS CS-Studio.app/Contents/Eclipse/configuration/diirt].
2017-05-23 09:13:24.899 WARNING [Thread 1] org.csstudio.diirt.util.core.preferences.DIIRTPreferences (getDIIRTHome) - The given path doesn't contain DIIRT configuration [/Users/claudiorosati/Desktop/ESS CS-Studio.app/Contents/Eclipse/configuration/diirt].
2017-05-23 09:13:24.899 CONFIG [Thread 1] org.csstudio.diirt.util.core.preferences.DIIRTPreferences (resolveAndVerifyDIIRTPath) - DIIRT home path resolved [before: platform:/config/diirt, after: /Users/claudiorosati/Desktop/ESS CS-Studio.app/Contents/Eclipse/configuration/diirt].
2017-05-23 09:13:24.899 WARNING [Thread 1] org.csstudio.diirt.util.preferences.BasePreferencePage (verifyAndNotifyWarning) - The given path doesn't contain DIIRT configuration [/Users/claudiorosati/Desktop/ESS CS-Studio.app/Contents/Eclipse/configuration/diirt].
```

mvitorovic, 2017-05-24:
@claudio-rosati Sorry, I was talking from memory. I remembered seeing always seeing problems when the file was missing, and I forgot it was not a stack trace.

claudio-rosati, 2017-05-24:
@mvitorovic 
This is the current situation: I cannot initialize the POJOs using DIIRTPreferences because they are created during the DIIRTPreferences initialization. So the rules to follow are the following:

- I you want DataSources be initialized only from  `plugin_customization.ini`, remove your `datasources.xml` file or make it empty (completely empty).

If `datasources.xml` is an empty data source (I mean an xml file with the `dataSources` element but nothing inside it), Default Data Source will be None, as you reported, because the hardcoded initialization of the POJOs is used. This is fine if you plan to edit the DataSources manually form the Properties dialog.

mvitorovic, 2017-05-24:
Thanks for checking this out.

Is it possible to turn off the warning once the 
org.csstudio.diirt.util.core.preferences.prefs file exists in the 
workspace?

> @mvitorovic <https://github.com/mvitorovic>
> This is the current situation: I cannot initialize the POJOs using
> DIIRTPreferences because they are created during the DIIRTPreferences
> initialization. [...]

---
Ta e-pošta je bila pregledana z Avast protivirusnim programom.
https://www.avast.com/antivirus

claudio-rosati, 2017-05-29:
Currently not.

If inside `org.csstudio.diirt.util.core.preferences.prefs` the reference to the DIIRT directory is bad or not all the expected files are found (e.g. `datasource.xml`) or succesfully loaded the warning appears.

Moreover, I'm not aware about the existance of the `org.csstudio.diirt.util.core.preferences.prefs` file because it is managed by the Eclipse preferences framework.

css-iter, 2017-06-28:
@claudio-rosati Hello, this is still giving us problems. Now with the empty or missing `datasources.xml` file the preferences are set correctly, but opening an OPI in runtime mode fails with the following Exception:

```
2017-06-28 12:07:20.320 CONFIG [Thread 1] org.csstudio.diirt.util.DiirtStartup (preWindowOpen) - DIIRT: preWindowOpen
2017-06-28 12:07:20.400 CONFIG [Thread 1] org.csstudio.diirt.util.DiirtStartup (preWindowOpen) - Setting diirt.home system property [/tmp/DIIRT2217398767488723201].
2017-06-28 12:07:20.400 CONFIG [Thread 1] org.csstudio.diirt.util.DiirtStartup (preWindowOpen) - Resetting the DIIRT configuration folder.
java.lang.NoClassDefFoundError: Could not initialize class org.diirt.datasource.PVManager
	at org.csstudio.diirt.util.DiirtStartup.preWindowOpen(DiirtStartup.java:55)
	at org.csstudio.utility.product.ApplicationWorkbenchWindowAdvisor.preWindowOpen(ApplicationWorkbenchWindowAdvisor.java:75)
	at org.csstudio.iter.css.product.ITERWorkbenchWindowAdvisor.preWindowOpen(ITERWorkbenchWindowAdvisor.java:41)
	at org.eclipse.ui.internal.WorkbenchWindow.fireWindowOpening(WorkbenchWindow.java:1652)
	at org.eclipse.ui.internal.WorkbenchWindow.setup(WorkbenchWindow.java:651)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:55)
	at org.eclipse.e4.core.internal.di.InjectorImpl.processAnnotated(InjectorImpl.java:966)
	at org.eclipse.e4.core.internal.di.InjectorImpl.inject(InjectorImpl.java:151)
	at org.eclipse.e4.core.internal.di.InjectorImpl.inject(InjectorImpl.java:118)
	at org.eclipse.e4.core.contexts.ContextInjectionFactory.inject(ContextInjectionFactory.java:73)
	at org.eclipse.ui.internal.Workbench.createWorkbenchWindow(Workbench.java:1531)
	at org.eclipse.ui.internal.Workbench.openWorkbenchWindow(Workbench.java:2659)
	at org.eclipse.ui.internal.Workbench.openWorkbenchWindow(Workbench.java:2651)
	at org.csstudio.opibuilder.runmode.RunModeService.createNewWorkbenchPage(RunModeService.java:152)
	at org.csstudio.opibuilder.actions.RunOPIAction.run(RunOPIAction.java:108)
	at org.eclipse.jface.action.Action.runWithEvent(Action.java:473)
	at org.eclipse.ui.actions.RetargetAction.runWithEvent(RetargetAction.java:235)
	at org.eclipse.jface.action.ActionContributionItem.handleWidgetSelection(ActionContributionItem.java:565)
	at org.eclipse.jface.action.ActionContributionItem.lambda$5(ActionContributionItem.java:436)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:86)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:5252)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1348)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:4522)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:4107)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$4.run(PartRenderingEngine.java:1121)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1022)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:150)
	at org.eclipse.ui.internal.Workbench$5.run(Workbench.java:693)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:610)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:148)
	at org.csstudio.utility.product.Workbench.runWorkbench(Workbench.java:103)
	at org.csstudio.startup.application.Application.startApplication(Application.java:265)
	at org.csstudio.startup.application.Application.start(Application.java:119)
	at org.csstudio.iter.css.product.ITERApplication.start(ITERApplication.java:83)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:196)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:388)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:243)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:673)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:610)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1519)
	at org.eclipse.equinox.launcher.Main.main(Main.java:1492)
Exception in thread "pool-4-thread-3" java.lang.NoClassDefFoundError: Could not initialize class org.diirt.datasource.PVManager
	at org.csstudio.simplepv.pvmanager.PVManagerPV.internalStart(PVManagerPV.java:253)
	at org.csstudio.simplepv.pvmanager.PVManagerPV.access$1(PVManagerPV.java:214)
	at org.csstudio.simplepv.pvmanager.PVManagerPV$5.run(PVManagerPV.java:363)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
[... cut ...]
If we put the XML to use CA into the `datasources.xml` the exception goes away. But then we lose the ability to set the preferences from `plugin_customization.ini`.

Also, the resolved path to the configuration folder gets saved in the workspace. This is again a problem for us, because the configuration is linked to the CODAC version, and upgrading the CODAC also installs new configuration and just changes the symbolic links. But since the code now saves the resolved target folders, the new CS-Studio does not get the latest configuration.

claudio-rosati, 2017-06-28:
I'm sorry I don't understand.
What do you mean exactly with: "opening an OPI in runtime mode"? (so I can try to reproduce the problem).
Moreover, it seems to me a strange problem because the error message is telling that the `org.diirt.datasource.PVManager` class cannot be initialised, but there are no hints about why. @shroffk @kasemir Have you an idea on how more information from PVManager can be logged?

css-iter, 2017-06-28:
You can open the OPI in runtime mode by pressing the right-most green icon (Looks like "play" button on the CD player) in the Toolbar

[image]

Or by right clicking on the OPI file in the Navigator and selecting `Open With... -> OPI Runtime`.

claudio-rosati, 2017-06-28:
OK, it is what I suposed to be.
So you can open end edit the OPI without eny problems, but when you run it the error appears. Am I correct?

css-iter, 2017-06-28:
Yes. That is correct.

css-iter, 2017-07-18:
Hello Claudio. Are you able to make any progress on this? Or do you have any schedule regarding this issue? Can you share where you currently are and what your plans look like?
Thank you in advance, Miha

claudio-rosati, 2017-07-18:
Hello Miha,
I'm sorry to not have yet answered to you. 
I've tried it now using `sim://` PVs and I cannot reproduce the problem. This is what I've done:

- I've located the `datasources.xml` my local version oc CS-Studio is using, edited it and deleted everything inside it (I made it a completely empty file);
- I've restarted CS-Studio (with no OPI opened) and verified that my preferences are still good (meaning that I'm using the ones stored in the preferences file)
- I've opened some of the BOY examples without the problem you have.
- I've opened a smapp BOY example using a text update on a true PV, againg without any troubles.

So I don't know how to get into the same problem you have. Moreover I don't understand why PVManager class  is not loaded without any specific error message... @shroffk have you an idea?

css-iter, 2017-07-18:
Thank you for your answer. Since you cannot reproduce it I'll start digging into this myself. I just didn't want to double the effort if you were working on this.

claudio-rosati, 2017-07-18:
Please let me know if/where you find something, and if can be the case for an hangout.

css-iter, 2017-08-14:
Ok. I found the problem. It's not in your code, but still, I'm don't understand how it worked for you.

The problem is in the org.diirt.datasource.DataSourceProvider.createDataSource()

Line 53: compsite.setConfiguration(readConfiguration(compsite, "datasources"));

The readConfiguration() returns 'null' in case it cannot read the datasources folder, and the setConfiguration() fails on receiving 'null'.

The solution is to return the default CompositeDataSourceConfiguration() instead of 'null'. This then correctly behaves by providing the plugin_customization.ini dictated .xml in TMP folder.

Anyhow, I'll contact @shroffk and propose a patch for the DIIRT. I'll reference this ticket for rationale.

claudio-rosati, 2017-08-14:
I'm very happy you've found the problem, and I apologize not being able to dedicate myself to the problem.

css-iter, 2017-08-14:
It's not a problem. I don't think this was in any way your task. I'm just reporting the findings here, because I think with this issue resolved, we can also close this ticket.