ControlSystemStudio/cs-studio #2088: Someone deleted the 4.4.x branch

berryma4, 2017-02-14:
It seems the 4.4.x branch has been deleted.
@willrogers, you made the last commit.  Can you push a new 4.4.x branch?

berryma4, 2017-02-14:
Also, is it possible to make branch rules in github? (like in bitbucket, or gitlab)

claudio-rosati, 2017-02-14:
Was my fault. I'm terribly sorry: I suposed to be on my forked version.

berryma4, 2017-02-14:
I found the branch permissions in the github settings, we should have set this a long time ago.
I'm protecting all frozen branches and requiring pull request to master.  We were doing this already, but the setting will help accidents from happening.

berryma4, 2017-02-14:
@shroffk , @kasemir , does anyone have an up to date 4.4.x?

kasemir, 2017-02-14:
I have origin/4.4.x from ~24 hours ago

willrogers, 2017-02-14:
I just pushed 4.4.x back - you might want to check it's in the right place but I think it is.

berryma4, 2017-02-14:
ce79cda884fee3d1d27bed74a2410d2860c7b153 should be the last commit.

kasemir, 2017-02-14:
I see eda3be3 as the last commit, and the origin/4.4.x that I had is the same that there's now.

berryma4, 2017-02-14:
@willrogers perfect!  Thank you so much ... I was just going to try `git reflog`.  But all is well again!
I will set the permissions on the 4.4.x  branch.

berryma4, 2017-02-14:
Thank you everyone!

shroffk, 2017-02-14:
right, the last commit I have matches wills and Kays not the ce79cda

berryma4, 2017-02-14:
Yes, sorry, my merge commit for that sha1 was the last.

berryma4, 2017-02-14:
Okay, I've protected released and master branches on cs-studio, thirdparty, maven-osgi-bundles, and product repos.
That should cover preventative measures.
Thank you again!

closed 2017-02-14 by berryma4

claudio-rosati, 2017-02-14:
Thank you and sorry again.