ControlSystemStudio/cs-studio #2101: Problems in my Jenckins pipeline

claudio-rosati, 2017-02-16:
It's about 24 hours or a bit more that the org.csstudio.logbook.olog.property.fault.test fails.

This is the error message:

[ERROR] Failed to execute goal org.eclipse.tycho:tycho-surefire-plugin:0.23.1:test (default-test) on project org.csstudio.logbook.olog.property.fault.test: Could not determine test framework provider. Available providers: [junit47, junit4, junit3]

Hav you any idea on hot to fix it?

claudio-rosati, 2017-02-17:
12:42:12 [INFO]                                                                         
12:42:12 [INFO] ------------------------------------------------------------------------
12:42:12 [INFO] Building org.csstudio.logbook.olog.property.fault.test 1.0.0-SNAPSHOT
12:42:12 [INFO] ------------------------------------------------------------------------
12:42:12 [INFO] 
12:42:12 [INFO] --- maven-clean-plugin:2.4.1:clean (default-clean) @ org.csstudio.logbook.olog.property.fault.test ---
12:42:12 [INFO] 
12:42:12 [INFO] --- tycho-packaging-plugin:0.23.1:build-qualifier (default-build-qualifier) @ org.csstudio.logbook.olog.property.fault.test ---
12:42:12 [INFO] The project's OSGi version is 1.0.0.201702171134
12:42:12 [INFO] 
12:42:12 [INFO] --- tycho-packaging-plugin:0.23.1:validate-id (default-validate-id) @ org.csstudio.logbook.olog.property.fault.test ---
12:42:12 [INFO] 
12:42:12 [INFO] --- tycho-packaging-plugin:0.23.1:validate-version (default-validate-version) @ org.csstudio.logbook.olog.property.fault.test ---
12:42:12 [INFO] 
12:42:12 [INFO] --- jacoco-maven-plugin:0.5.3.201107060350:prepare-agent (default) @ org.csstudio.logbook.olog.property.fault.test ---
12:42:12 [INFO] tycho.testArgLine set to -javaagent:/home/jenkins/workspace/ESS-CS-Studio/.m2/org/jacoco/org.jacoco.agent/0.5.3.201107060350/org.jacoco.agent-0.5.3.201107060350-runtime.jar=destfile=/home/jenkins/workspace/ESS-CS-Studio/cs-studio/applications/logbook/logbook-plugins/org.csstudio.logbook.olog.property.fault.test/../target/jacoco.exec,includes=*
12:42:12 [INFO] 
12:42:12 [INFO] --- maven-resources-plugin:2.4.3:resources (default-resources) @ org.csstudio.logbook.olog.property.fault.test ---
12:42:12 [INFO] Using 'UTF-8' encoding to copy filtered resources.
12:42:12 [INFO] skip non existing resourceDirectory /home/jenkins/workspace/ESS-CS-Studio/cs-studio/applications/logbook/logbook-plugins/org.csstudio.logbook.olog.property.fault.test/src/main/resources
12:42:12 [INFO] 
12:42:12 [INFO] --- tycho-compiler-plugin:0.23.1:compile (default-compile) @ org.csstudio.logbook.olog.property.fault.test ---
12:42:12 [INFO] Compiling 4 source files to /home/jenkins/workspace/ESS-CS-Studio/cs-studio/applications/logbook/logbook-plugins/org.csstudio.logbook.olog.property.fault.test/target/classes
12:42:12 [INFO] 
12:42:12 [INFO] --- maven-resources-plugin:2.4.3:testResources (default-testResources) @ org.csstudio.logbook.olog.property.fault.test ---
12:42:12 [INFO] Using 'UTF-8' encoding to copy filtered resources.
12:42:12 [INFO] skip non existing resourceDirectory /home/jenkins/workspace/ESS-CS-Studio/cs-studio/applications/logbook/logbook-plugins/org.csstudio.logbook.olog.property.fault.test/src/test/resources
12:42:12 [INFO] 
12:42:12 [INFO] --- tycho-source-plugin:0.23.1:plugin-source (plugin-source) @ org.csstudio.logbook.olog.property.fault.test ---
12:42:12 [WARNING] bundle localization file /home/jenkins/workspace/ESS-CS-Studio/cs-studio/applications/logbook/logbook-plugins/org.csstudio.logbook.olog.property.fault.test/plugin.properties not found
12:42:12 [WARNING] Bundle-Vendor header not found in /home/jenkins/workspace/ESS-CS-Studio/cs-studio/applications/logbook/logbook-plugins/org.csstudio.logbook.olog.property.fault.test/META-INF/MANIFEST.MF, fallback to 'unknown' for source bundle
12:42:12 [INFO] Building jar: /home/jenkins/workspace/ESS-CS-Studio/cs-studio/applications/logbook/logbook-plugins/org.csstudio.logbook.olog.property.fault.test/target/org.csstudio.logbook.olog.property.fault.test-1.0.0-SNAPSHOT-sources.jar
12:42:12 [INFO] 
12:42:12 [INFO] --- target-platform-configuration:0.23.1:target-platform (default-target-platform) @ org.csstudio.logbook.olog.property.fault.test ---
12:42:12 [INFO] 
12:42:12 [INFO] --- tycho-packaging-plugin:0.23.1:package-plugin (default-package-plugin) @ org.csstudio.logbook.olog.property.fault.test ---
12:42:12 [INFO] Building jar: /home/jenkins/workspace/ESS-CS-Studio/cs-studio/applications/logbook/logbook-plugins/org.csstudio.logbook.olog.property.fault.test/target/org.csstudio.logbook.olog.property.fault.test-1.0.0-SNAPSHOT.jar
12:42:12 [INFO] 
12:42:12 [INFO] --- tycho-p2-plugin:0.23.1:p2-metadata-default (default-p2-metadata-default) @ org.csstudio.logbook.olog.property.fault.test ---
12:42:12 [INFO] No baseline version MavenProject: org.csstudio:org.csstudio.logbook.olog.property.fault.test:1.0.0-SNAPSHOT @ /home/jenkins/workspace/ESS-CS-Studio/cs-studio/applications/logbook/logbook-plugins/org.csstudio.logbook.olog.property.fault.test/pom.xml
12:42:12 [INFO] 
12:42:12 [INFO] --- tycho-surefire-plugin:0.23.1:test (default-test) @ org.csstudio.logbook.olog.property.fault.test ---
12:42:14 [INFO] ------------------------------------------------------------------------
12:42:14 [INFO] Reactor Summary:

...
12:42:14 [INFO] org.csstudio.logbook.olog.property.fault.test ..... FAILURE [2.412s]
...

12:42:18 [ERROR] Failed to execute goal org.eclipse.tycho:tycho-surefire-plugin:0.23.1:test (default-test) on project org.csstudio.logbook.olog.property.fault.test: Could not determine test framework provider. Available providers: [junit47, junit4, junit3] -> [Help 1]

claudio-rosati, 2017-02-17:
@berryma4 @willrogers @kasemir @shroffk 

Have you the same problem?

willrogers, 2017-02-17:
Yes, I get the same problem.  I think we've missed this because the build wasn't failing correctly for the past few days.

claudio-rosati, 2017-02-17:
OK, thank you. I'll wait for a fix.

berryma4, 2017-02-17:
@willrogers removed the test.  @shroffk can add it back when it is passing.

closed 2017-02-17 by berryma4

shroffk, 2017-02-20:
@willrogers thank you for resolving the checkstyles...I was halfway though fixing them myself before I noticed your fixes.

@willrogers I guess due to my target setup I was not seeing the same errors as seen by you and claudio. It seems like your fix for the imported bundles should resolve it.