ControlSystemStudio/cs-studio #2102: Inspect Waveform Viewer with log scale

shroffk, 2017-02-16:
We have some data that is really hard to interrupt without being able to customize the axes since the data spans 3+ orders of magnitude.
It would be useful to have an option (context menu action) to allow the YAxis to to toggled between linear and logarthmin

kasemir, 2017-02-16:
We have the same requirement, but only for some data.
If we add that as a toolbar or context menu item, people click that for data that contains values <= 0 and then complain why there's no log(-1).

So we use a button that writes to loc://log_axis(0) and then have a rule on the plot that sets the axis' log scale option from that PV
-> People can toggle log axis on/off but only for those displays where we decided to add the button because it makes sense for that data.

kasemir, 2017-02-16:
Rule looks like this for BOY:
```
 <rule name="Log Rule" prop_id="axis_1_log_scale" out_exp="false">
          <exp bool_exp="pv0">
            <value>true</value>
          </exp>
          <pv trig="true">loc://log_axis(0)</pv>
</rule>
```

For the Display Builder the property ID changes:
```
<rule name="Log Scale Rule" prop_id="y_axes[0].log_scale" out_exp="false">
          <exp bool_exp="pv0">
            <value>true</value>
          </exp>
          <pv_name>loc://log_axis(0)</pv_name>
</rule>
```

shroffk, 2017-02-16:
Ok, 
>> complain why there's no log(-1). 
hmmm....that is just wonderful

I understand your fix for BOY and display builder, however my primary use case comes from people looking at this data from the archiver using the databrowser inspect waveform.

Could we just add the action to the inspect waveform view, if people try to plot negative values...then too bad blame maths.

shroffk, 2017-02-16:
This way the action is only present in the inspect waveform view...and not included in all the other uses of RTplot

shroffk, 2017-02-16:
@kasemir would this be acceptable?

kasemir, 2017-02-16:
Yes, adding that to the waveform view is OK.
I'm fine with your #2102, make a pull request, and I'll add that to the updated data browser.

willrogers, 2017-02-16:
We have a few more user requests for the waveform view.  I'll look at what Kunal's done and consider how we might do some of ours.

shroffk, 2017-02-16:
FYI, the ```refresh()``` after the resetting axis type does not really result in the refresh. You have to resize of move the graph for the actual redraw to happen.
In the past I have gotten around this by calling the ```pack()``` but am hoping we find a better solution to this.

kasemir, 2017-02-16:
Try adding
```
 plot.requestUpdate();
```

The `refresh` is just an SWT widget refresh, not updating the underlying plot image.
`requestUpdate()` re-creates the plot image.

--

I've added this to the updated databrowser.

shroffk, 2017-02-16:
Perfect...it works now

closed 2017-02-21 by berryma4