ControlSystemStudio/cs-studio #2104: Offline build problem

willrogers, 2017-02-16:
We have had a real problem with our offline build.

After a lot of debugging, we track it down to the plugin `javax.xml.jre` from the EFX repo.  The problem appears to be that it does approximately the same job as `javax.xml` which is in Orbit - that is, repackaging `javax.xml` and `org.xml.sax` - at the same version 2.0.2.  For very specific reasons this causes us a problem which probably hits no-one else as most builds are online.

If I understand correctly, we could filter this plugin out of the build completely and suffer no ill-effects.  Do you mind if I try this?

willrogers, 2017-02-16:
I learnt a bunch more about Maven and Tycho while investigating this, and I still haven't got to the bottom of it.

I have removed the plugin from where it was causing a problem, in Save Restore.  But alternatively we could remove it from both `core` and `applications` completely.  We might yet find this solution causes another problem but I hope not.

berryma4, 2017-02-21:
Can we close this now?

willrogers, 2017-02-22:
Yes, this is now working for us.

closed 2017-02-22 by willrogers