ControlSystemStudio/cs-studio #2111: Missing preference page

claudio-rosati, 2017-02-17:
Eclipse has teh **Startup and Shutdown** preference page.

[image]

Inside it the **Workspaces** sub-page allows managing the visited workspaces and define the behaviour at startup.

[image]

Moreover, the **Workspace** page is linking the **Startup and Shutdown** one just below the header.

[image]

CS-Studio has a problem, not being able to find the org.eclipse.ui.preferencePages.Startup. How this can be solved? I've tried googling without success.

[image]

jbobnar, 2017-02-19:
- If you want to use the exact preference page that Eclipse is using add the plugin org.eclipse.ui.ide.application to your product.
- Alternatively you can create a plugin that provides the definition of the extension point org.eclipse.ui.preferencePages and use the above name for its ID.

claudio-rosati, 2017-02-20:
Your first point is exactly what I've tried to do using the teature editor on my product's feateures, but the editor didn't show me the org.eclipse.ui.ide.application plugin.

So after your suggestion, I've tried to add the plugin editing the features' XML file, built the product with Maven/tycho and now the property page appeared.

Thank you

closed 2017-02-20 by claudio-rosati