ControlSystemStudio/cs-studio #2132: XYGraph plotted data disappear after some time when scrolling is disabled

rosarp, 2017-02-23:
Even though the graph is "paused", live data continue to be stored in the ring
buffer and make the plotted data disappear.

It is requested to:

- keep the plotted data when the scrolling is disabled
- when the scrolling is enabled again, a new request to the archive repository
shall be sent to refresh the data

kasemir, 2017-02-23:
Pausing the scroll should not pause collecting the live data.

People to setup the live data buffer to for example hold 24 hours of a given sample. They then want to zoom/pan around in that time range, eventually re-enable the scroll and have the recent data. Stopping the live data collection would be wrong.

If you stop the scrolling and the live data buffer then rolls out of the visible range, there are two options:

 * Increase the live data buffer size
 * Check the automatic_history_refresh mechanism. It usually fetches archived data to fill the gap between archived data and the start of the live data buffer as that fills up during scrolling. Maybe it needs a tweak to do the same when not scrolling.

kasemir, 2017-02-23:
Sorry, I notice now that you were talking about the XYGraph, not the data browser widget.
So go ahead with that.

PR #2133 (2017-02-24): XYGraph plotted data disappear after some time when scrolling is disabled
Fixes #2132
closed 2017-02-24 by berryma4