ControlSystemStudio/cs-studio #2134: diirt config:  file in temp is empty

berryma4, 2017-02-23:
I have a datasource (scan://) that I'm trying to load up.  When I select the diirt folder, it seems a file is copied to Temp/DIIRT[0-9]+/datasources/scan/scan.xml, but then this file is empty.  So the datasource fails.

berryma4, 2017-02-23:
```java
2017-02-23 15:05:33.124 SEVERE [Thread 1] org.diirt.datasource.ConfigurableDataSourceProvider (createInstance) - Couldn't load DIIRT_HOME/datasources/scan/file.xml
java.lang.NullPointerException
	at org.diirt.util.config.Configuration.getFile(Configuration.java:79)
	at org.diirt.util.config.Configuration.getFileAsStream(Configuration.java:91)
	at org.diirt.datasource.ConfigurableDataSourceProvider.createInstance(ConfigurableDataSourceProvider.java:87)
	at org.diirt.datasource.ConfigurableDataSourceProvider.createInstance(ConfigurableDataSourceProvider.java:74)
	at org.diirt.datasource.CompositeDataSource.retrieveDataSource(CompositeDataSource.java:232)
	at org.diirt.datasource.CompositeDataSource.connectRead(CompositeDataSource.java:177)
	at org.diirt.datasource.PVDirector.connectReadExpression(PVDirector.java:132)
	at org.diirt.datasource.PVReaderConfiguration.prepareDecoupler(PVReaderConfiguration.java:166)
	at org.diirt.datasource.PVReaderConfiguration.maxRate(PVReaderConfiguration.java:131)
	at org.diirt.datasource.PVConfiguration.asynchWriteAndMaxReadRate(PVConfiguration.java:146)
	at org.csstudio.diag.pvmanager.probe.PVManagerProbe.setPVFormula(PVManagerProbe.java:371)
	at org.csstudio.diag.pvmanager.probe.PVManagerProbe.createPartControl(PVManagerProbe.java:277)
	at org.eclipse.ui.internal.e4.compatibility.CompatibilityPart.createPartControl(CompatibilityPart.java:151)
	at org.eclipse.ui.internal.e4.compatibility.CompatibilityView.createPartControl(CompatibilityView.java:143)
	at org.eclipse.ui.internal.e4.compatibility.CompatibilityPart.create(CompatibilityPart.java:341)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:56)
	at org.eclipse.e4.core.internal.di.InjectorImpl.processAnnotated(InjectorImpl.java:898)
	at org.eclipse.e4.core.internal.di.InjectorImpl.processAnnotated(InjectorImpl.java:879)
	at org.eclipse.e4.core.internal.di.InjectorImpl.inject(InjectorImpl.java:121)
	at org.eclipse.e4.core.internal.di.InjectorImpl.internalMake(InjectorImpl.java:345)
	at org.eclipse.e4.core.internal.di.InjectorImpl.make(InjectorImpl.java:264)
	at org.eclipse.e4.core.contexts.ContextInjectionFactory.make(ContextInjectionFactory.java:162)
	at org.eclipse.e4.ui.internal.workbench.ReflectionContributionFactory.createFromBundle(ReflectionContributionFactory.java:104)
	at org.eclipse.e4.ui.internal.workbench.ReflectionContributionFactory.doCreate(ReflectionContributionFactory.java:73)
	at org.eclipse.e4.ui.internal.workbench.ReflectionContributionFactory.create(ReflectionContributionFactory.java:55)
	at org.eclipse.e4.ui.workbench.renderers.swt.ContributedPartRenderer.createWidget(ContributedPartRenderer.java:129)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.createWidget(PartRenderingEngine.java:971)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeCreateGui(PartRenderingEngine.java:640)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$1.run(PartRenderingEngine.java:525)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:42)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.createGui(PartRenderingEngine.java:509)
	at org.eclipse.e4.ui.workbench.renderers.swt.ElementReferenceRenderer.createWidget(ElementReferenceRenderer.java:69)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.createWidget(PartRenderingEngine.java:971)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeCreateGui(PartRenderingEngine.java:640)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeCreateGui(PartRenderingEngine.java:746)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.access$0(PartRenderingEngine.java:717)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$2.run(PartRenderingEngine.java:711)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:42)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.createGui(PartRenderingEngine.java:695)
	at org.eclipse.e4.ui.workbench.renderers.swt.StackRenderer.showTab(StackRenderer.java:1306)
	at org.eclipse.e4.ui.workbench.renderers.swt.LazyStackRenderer.postProcess(LazyStackRenderer.java:103)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeCreateGui(PartRenderingEngine.java:658)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeCreateGui(PartRenderingEngine.java:746)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.access$0(PartRenderingEngine.java:717)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$2.run(PartRenderingEngine.java:711)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:42)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.createGui(PartRenderingEngine.java:695)
	at org.eclipse.e4.ui.workbench.renderers.swt.SWTPartRenderer.processContents(SWTPartRenderer.java:71)
	at org.eclipse.e4.ui.workbench.renderers.swt.SashRenderer.processContents(SashRenderer.java:151)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeCreateGui(PartRenderingEngine.java:654)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeCreateGui(PartRenderingEngine.java:746)
[... cut ...]
	at org.eclipse.equinox.launcher.Main.main(Main.java:1488)
Caused by: org.xml.sax.SAXParseException; lineNumber: 1; columnNumber: 1; Premature end of file.
	at com.sun.org.apache.xerces.internal.parsers.DOMParser.parse(DOMParser.java:257)
	at com.sun.org.apache.xerces.internal.jaxp.DocumentBuilderImpl.parse(DocumentBuilderImpl.java:339)
	at javax.xml.parsers.DocumentBuilder.parse(DocumentBuilder.java:121)
	at org.csstudio.scan.diirt.datasource.ScanDataSourceConfiguration.read(ScanDataSourceConfiguration.java:46)
	... 124 more
```

claudio-rosati, 2017-02-27:
If I remember correctly, xml files are not copied. Instead they are read creating a corresponding java object, then this object is used to save it into a new xml file.

So, to solve this problem (I would like to do it together with #2086) I need to know:

- which other XML files other then ca.xml must be managed?
- where can I find the documentation (or at least a full example) of them?
- must a proper preference page be set up for them too (as a child of Data Source)?

Thank you,
Claudio

berryma4, 2017-02-27:
I don't plan on having a preference ui for this.  I'm not sure why it is
creating the directory and file in temp, I rather it just read the one in
the diirt config repo for this datasource.

claudio-rosati, 2017-02-27:
The new mecanism will always (re-)create the diirt files in a temporary folder.
Can I have a copy of this file or the specification of this XML file?

berryma4, 2017-02-27:
```xml
<?xml version="1.0" encoding="UTF-8" standalone="yes" ?>
<scan version="1">
  <servers pollInterval="2" timeout="10">
    <server name="ecrscan" url="http://localhost:4810/">
    </server>
  </servers>
</scan>
```

claudio-rosati, 2017-02-28:
I understand and apologize to not have done it yet... meetings

The same thing will happen to all other data sources having xml files into the diirt directory. 

Some small piece of code must be written. I'll do it tomorrow. Please attach all xml involved so I'll handle them in a single commit.

claudio-rosati, 2017-02-28:
Alternatively, I can simplify things just copying everything that has no corresponding preference page.

berryma4, 2017-02-28:
That seems okay.
Thank you!

berryma4, 2017-03-01:
@claudio-rosati thank you so much!

closed 2017-03-01 by berryma4

claudio-rosati, 2017-03-02:
@berryma4 can thisissue be closed?