ControlSystemStudio/cs-studio #2144: Polar plotting

claudio-rosati, 2017-03-08:
Here another question for a colleague at ESS:

> We have data points that would need to be plotted in polar coordinate system. Points span from -PI to +PI and plotting them in XY graph does not fit our needs.
Can CSS plot in polar coordinate system?

@kasemir Is that possible in Display Builder?

kasemir, 2017-03-09:
I'm not aware of any special support for polar coordinates in edd/dm, medm, dm2k, edm, BOY, and the display builder also doesn't have that.
The SNS LLRF for example has plots of RF phases, range +-180 degrees, in EDM. They're plain X/Y plots where the Y axis is valued -180 .. +180. Means that phases around 180 tend to jump between the top and bottom of the plot.
There are also some visualizations in BOY where an "arrow" (2-point polyline with arrow on one end) is rotated via a script/rule, but that's only useful for displaying a single angle, not a waveform of angles over time.
Are there suggestions what it should look like?

claudio-rosati, 2017-03-09:
@kasemir 
Here the answer from my colleagues:

We have the same use case as Kay mentioned; we are experiencing exactly the same behavior of phases jumping from 180 to -180 degrees due to noise in the signal. Plot in XY graph can look messy and might suggest that there are problems, when there are not. OTOH, polar plot show the data in their 'natural' form and make lot more sense.

We want to plot all the angles of a single pulse (thousands of angles) from a waveform PV.
We will also do an average of all angles per pulse (get scalar per pulse) and collect those in a waveform PV and then plot them in the similar fashion as above.
Along with the angle should also go amplitude waveform of the same length.
The widget would thus take two PV waveforms (angle, amplitude) and generate a polar plot using those. If amplitudes waveform is not specified plot all angles with the same amplitude (eg. 1).
BPM system is the one that would benefit from this type of plot.

[image]

So the decision was to add a new widget type in the list of widgets I have to implement for Display Builder.

kasemir, 2017-03-10:
We do create something similar to your polar plot example in EDM. I'll have to get screenshots when returning from travel. Trouble is that you can't compare that with the amplitude over time because there's no time axis in the polar plot.

Another option for the angle over time might be a basic XY type of plot where the autoscale for the Y axis is aware of handling degrees. So if the value is around 180 degrees, it would place 180 degrees at the "0" location. Still, during cavity fill the phase can rotate quite a bit, so there's still a chance of 'wrapping around'.

closed 2018-06-12 by berryma4