ControlSystemStudio/cs-studio #2149: BOY transparent rectangle shouldn't intercept clicks

willrogers, 2017-03-10:
Background: quite often you can have a screen with a transparent rectangle used as a border or outline.  Sometimes there could be buttons or readbacks behind this.

If the rectangle has no actions or tooltip, you should be able press buttons or get tooltips from lower elements.

Realistically, this is more likely to happen when converting screens from EDM (that's how I found it), but I still think the described behaviour is sensible.  It would be sensible for other shapes too, but it seems less likely to happen.

The label widget already has this behaviour, so I can copy from that.

claudio-rosati, 2017-03-10:
Isn't better having a new boolean property telling if the object (transparent or not) will intercept mouse events?

willrogers, 2017-03-10:
Hm... two points:

* The logic we're going for here is: if you can see a button, you should probably be able to click it.  Anything else is a bit surprising.  Would the option really help when creating screens?
* Also, your suggestion doesn't actually help my actual problem.  When we're doing EDM conversion, it's difficult to tell what is in front of what.  I suppose we could make all rectangle un-clickable when converting if they don't have actions attached to them.

claudio-rosati, 2017-03-13:
I understand. My was more a proposal for a general approach on mouse events..

kasemir, 2017-03-13:
I've implemented that for the display builder because of the EDM legacy: Transparent rectangles pass all mouse events through. This also means you can no longer open the widget information dialog for such rectangles at runtime.

Fundamentally, I think it would be better to have such rectangles 'behind' the buttons, or use a 'group' to, well, group widgets instead of lines or rectangles. Still, making these rectangles transparent to mouse clicks seems necessary for porting EDM files.

willrogers, 2017-03-13:
I'm happy for you to choose the 'correct' behaviour for display builder, but it seems you also have a view to EDM compatibility.

This change makes things much easier for us when converting from EDM.

closed 2017-04-04 by willrogers