ControlSystemStudio/cs-studio #2156: Displaying unit in TextInput and TextUpdate (BOY & DB)

claudio-rosati, 2017-03-13:
Hello all,

this is a feature request from colleagues at ESS (for BOY & Display Builder - @kasemir).

Currently it is only possible to decide if units (coming from PV) must be displayed or not in TextInput and TextUpdate widgets. The request is for being possible to display units passed by the user. This means:

- having a new string property for the unit,
- changing the current Show Unit property from a boolean to a 3-values enumeration (OFF, FROM PV, FROM PROPERTY - or something similar).

I've asked the colleagues why not just using a label close to the TextInput/TextUpdate for the unit, and the answer was because the global rendering (alignment with the value, font, position) will be different requiring a lot of graphical work to fix it, moreover the alarm border will be another issue too (requiring a rectangle below the value and the unit for the alarm border, and disabling alarm border on value and unit).

kasemir, 2017-03-13:
Channel Access and PVAccess already include the units. The DBE_PROPERTY support was added to support changing units at runtime.

If a PV doesn't include units, that needs to be fixed at the IOC level because the units are not just for BOY or display builder. They're also for the archive, alarms, other tools.
If it's for some reason impossible to add the units to the IOC, for example because it's a LabVIEW based CA server handled by somebody who tells you that LabVIEW is so easy to use, yet he cannot figure out how to provide units (or PREC, MDEL, valid time stamp ..), then there's still the option to add a soft IOC:

````
record(ai, "TheOneYouShow")
{
  field(INP, "TheOneWithoutUnits CP")
  field(EGU, "Units")
}
````

You could also use a Label widget with a rule or script to update the text based on a value PV and units from either a constant or other PV. But that would be like using a rule to decode a number into strings, instead of having an enum PV.

alex-soderqvist, 2017-03-14:
My view is purely aesthetic and practical. Of course the database should be correct, there are just a lot of scenarios when it is impossible or very inconvenient to do so.

The PV EGU may be incoherent, because they are coming from different databases.

Example:
seconds
Seconds
Sec
sec
s

centigrades
°C
C

The OPI could potentially be (re)designed by someone not even familiar with IOCs and databases.

It would be very convenient to be able to set this. In these situations you are just likely not do anything at all.

closed 2018-06-12 by berryma4