ControlSystemStudio/cs-studio #2177: Diirt PV names not resolved in tooltips

willrogers, 2017-04-04:
It is useful that BOY shows PV name in a tooltip and you can copy the value, either with middle-click or the context menu.

However, when working with Diirt expressions, the actual PV text may not be particularly helpful:

* `=pv(concat('loc://DID_9_PVName', ".HIHI"))`

when what I am actually viewing is 

* `SR04A-PC-Q3D-08:SETI.HIHI`

Is there any way to extract the actual PV from this in Diirt?  I looked once and found that this was buried fairly deeply, but perhaps someone who knows more could do it.

@shroffk @berryma4 @kasemir

shroffk, 2017-04-04:
As far as I am aware this is not possible without the use of scripts.

It would be nice if we could set
```tooltip: =concat('loc://DID_9_PVName', ".HIHI")```

kasemir, 2017-04-04:
I'm using scripts or rules when a pv_name property needs to change at runtime. That way the name used for the context menu and tooltip is updated accordingly.

kasemir, 2017-04-04:
Doesn't that only set the tooltip, while the PV from the context menu stays the same? So copy-to-clipboard and then paste into another tool won't get the PV name.

berryma4, 2017-04-04:
Ah yes, the context menu.  For the context menu, it seems we would need to come up with another way, considering someone might want a formula as the context (pvmanager wouldn't know when to use = x+ y or expand =pv(x) ).