ControlSystemStudio/cs-studio #2179: Snapshot not working on MacOS X

claudio-rosati, 2017-04-06:
Hello all,

the ESS product works fine taking screen snapshots on CentOS and Windows, not on MacOS X where the "captured" image is completely white.

Have someone noticed the same problem on MacOS X? Have I missed something?

Claudio

kasemir, 2017-04-06:
Which specific screen snapshot?
I've tried these:
 * DataBrowser3 context menu "Save Snapshot to file": Works. Same code is used to the the email and logbook snapshot, which are fine as well.
 * Display Builder runtime context menu "Save Snapshot to file", email, logbook: Fine.
 * BOY runtime: Context menu has no direct save-to-file. Can get snapshot via print (.. to file), send email, submit logbook entry: Fine.

In all these examples, the email and logbook dialogs offer "Screen shot" and "CSS Window" buttons, which succeed in obtaining the complete or noly-CSS-window snapshot.

I do remember getting blank images while developing the JavaFX based screen capture that's underneath the display builder snapshot support. Some image types would not work. Forgot the details, can only tell that I ended up writing an ARGB image as *.png.

kasemir, 2017-04-06:
Info from hangout: It's about org.csstudio.utility.screenshot.
Could change this into issue for org.csstudio.utility.screenshot not working on Mac OS X.
In principle, that should be easy to fix:
The Elog and EMail submit dialogs contain SWT code for getting the CSS Window or complete desktop image, so put that into org.csstudio.utility.screenshot.

claudio-rosati, 2017-04-06:
I'll look at that. 
Thank you