ControlSystemStudio/cs-studio #2199: Catch zero-length PVs

kasemir, 2017-04-27:
vtype.pv should throw exception for zero-length PV name requests because they might crash the CA gateway, http://www.aps.anl.gov/epics/tech-talk/2017/msg00714.php

kasemir, 2017-04-27:
vtype.pv no longer passes zero-length PV names on to JCA.

Found that the EPICS PV Tree would request empty PV when no PV name is entered.
More tools requesting empty PVs will now be caught because of vtype.pv update.

anjohnson, 2017-04-27:
The underlying caj client library should be catching empty PV names itself and not passing them on, but evidently it doesn't. The libCa implementation returns `ECA_BADSTR` from `ca_create_channel()` when handed an empty PV name, so C-based clients can't cause this problem.

kasemir, 2017-04-28:
Agree that CAJ should also catch empty PV names, for example 
```
diff -r 4e82abf3d65b src/com/cosylab/epics/caj/CAJChannel.java
--- a/src/com/cosylab/epics/caj/CAJChannel.java	Mon Jun 29 21:01:06 2015 +0200
+++ b/src/com/cosylab/epics/caj/CAJChannel.java	Fri Apr 28 09:22:53 2017 -0400
@@ -186,6 +186,8 @@
	protected CAJChannel(CAJContext context, int channelID, String name,
			ConnectionListener listener, short priority) throws CAException
	{
+		if (name == null  ||  name.isEmpty())
+			throw new CAException("Empty channel name");
		this.context = context;
		this.channelID = channelID;
		this.name = name;
```

Contacted Matej.