ControlSystemStudio/cs-studio #2204: Minimize the editor when OPI is opened as standalone window

lcavalli, 2017-05-12:
I need to open an OPI file as a standalone windows, hiding the editor and leaving only the OPI file visible.

To open the file I can used the "Position" macro (btw, NEW_SHELL option seems undocumented):
`cs-studio --launcher.openFile "<path_to_opi>/file.opi Position=NEW_SHELL"`

I added an additional macro to be used together with Position=NEW_SHELL to automatically hide the Workbench window:
`cs-studio --launcher.openFile "<path_to_opi>/file.opi Position=NEW_SHELL, MinimizeEditor=true"`

I'm going to open a pull request with the feature added, alongside the NEW_SHELL option documented in help.

willrogers, 2017-05-12:
Hi,

Is this when you launch CS-Studio for the first time, or when you launch another screen when it is already running, or both?

I had some ideas about being able to 'minimise' CSS to the tray, in a similar way to Spotify or Skype.  This might allow people who are not interested in the main window to keep it out of the way.

Cheers,
Will

lcavalli, 2017-05-13:
Hi Will,

my idea was to make the feature available every time a new OPI is opened as standalone window with the MiminimzeEditor macro set, but my patch doesn't work in all cases.
1. First cs-studio run executed from command line: WORKS
2. Second cs-studio run executed from command line (with previous still running): DOESN'T WORK
3. OPI file opened from action button (from a standalone windows): WORKS
4. OPI file opened from action button (not from a standalone windows): WORKS (by minimizing the windows containing the calling action button)

My use case is covered by points 1 and 2 (I would like to have a desktop icon to open the main GUI panel as a standalone window, if someone click the icon a second time with cs-studio still minimized I will fall lin point 2). I think I can fix point 4 by calling getWorkbenchWindows() and find a way to catch main CSS window (with getParent on its Shell returning null?).

Any advice to improve the patch to fix point 2?

Thanks!

willrogers, 2017-05-17:
I can't see why point 2 isn't working without giving it a try.

Does the new window open, but the main window isn't minimised?  Or is the second window not opened?  You are silently swallowing exceptions - you should at least log them in case there's some information.

lcavalli, 2017-05-17:
Correct, I modified the code to log an error message in case of exception (I will commit it soon to my fork).
Nonetheless no error messages are printed.

I added a console information message before and after the minimizing code. On second command line execution the workbench pops out together with my OPI as a standalone window and **then** it prints the messages on the console, but the editor is not minimized.

willrogers, 2017-05-17:
My guess is that is something to do with the automatic behaviour of opening a file:  by default it maximises CS-Studio because this is what you'd usually want when opening a file in Eclipse.

Still not sure why it's different between first and subsequent opening.

willrogers, 2017-07-05:
Hi @lcavalli,

I would like to propose an idea.  In the case that a user wants to keep the main window out of the way, he could minimise it to the 'tray' (it's the C icon here):

[image]

If CS-Studio is minimised in this way, opening a standalone window will not maximise the main window.  You can get the main window back if you click on the icon.  You can exit CS-Studio completely by right-click -> Exit on the icon.

Would this meet your requirements?  This behaviour would be enabled by a preference.

Cheers,
Will

lcavalli, 2017-07-05:
@willrogers, absolutely yes, this would meet my requirements given that CS-Studio can start minimized in system tray without any manual operation by the operator (e.g. with a command line option). Will this be possible?

willrogers, 2017-07-05:
It does seem to be possible.  @berryma4 @kasemir @shroffk is this something that would be acceptable?  I think it could be disabled by default using a preference.

There are some edge behaviour cases, but this applies to your examples too:

- What if you right-click and select 'databrowser' or 'probe'?  At the moment the main window remains hidden, so you don't see anything happen
- What if something causes a pop-up?  At the moment, the main window __does__ appear with the pop-up on top

berryma4, 2017-09-05:
close?

lcavalli, 2017-09-05:
Forgot to close this, sorry.

closed 2017-09-05 by lcavalli