ControlSystemStudio/cs-studio #2206: Popup message are displayed behind the running OPI in Full Screen mode

utzeln, 2017-05-12:
We have experienced on Red Hat 6 with GNOME 2 many issues with popup messages requesting a user validation but displayed behind the running OPI in Full Screen mode.

We hacked JFace plugin - https://github.com/css-iter/org.csstudio.iter/tree/master/plugins/org.eclipse.jface.

With Red Hat 7 and GNOME 3 we still have the same issue but the hack on JFace do not work.

berryma4, 2017-05-12:
@shroffk didn't you mention something similar?

shroffk, 2017-05-12:
Yes,
I have had this issue since the move to the 4.x series. I have this issue on gdk3 on debian...
What was the hack in your jface?

utzeln, 2017-05-12:
We made a copy of the official org.eclipse.jface plugin and the only change from the original is the file org.eclipse.jface.dialogs.Dialog.java. In the file Dialog.org in the root of the project you will find the original file and in the src folder you can find the patched file. 

The last known working plugin was part of the eclipse luna.

boryto, 2017-06-05:
The fix was creating a dummy shell over existing shell, but the fix stooped working with the Gnome-3, GTK-3.  The bug was even worse on GTK-3 as the CSS stooped working completely.

Bit it seems there is progress in fixing this bug on Eclipse Oxygen release. 

We used used the latest swt.gtk (https://git.eclipse.org/c/platform/eclipse.platform.swt.binaries.git/tree/bundles/org.eclipse.swt.gtk.linux.x86_64) library (that comes from  Eclipse 4.7_RC2) over the existing swt.gtk library. 
The library resolved most of the issues. 

The only problem that remained was the Send - email popup. The cause was the non-modal style used setShellStyle(style & ~SWT.APPLICATION_MODAL);

We fixed the issue with passing the empty shell to the Send e-mail dialog (in-full screen mode), so the dialog is fully detached from the CS studio. 

This is the same way as it works in the "Create long entry" popup.

berryma4, 2017-09-05:
close?