ControlSystemStudio/cs-studio #2210: Builde Failure

claudio-rosati, 2017-05-29:
Hello,

I'm unable to build the master branch after the last merges...

```
[ERROR] Failed to execute goal org.eclipse.tycho:tycho-compiler-plugin:1.0.0:compile (default-compile) on project org.csstudio.utility.product: Compilation failure: Compilation failure:
[ERROR] /Users/claudiorosati/Projects/GitHub/cs-studio/core/utility/utility-plugins/org.csstudio.utility.product/src/org/csstudio/utility/product/ApplicationWorkbenchAdvisor.java:[156]
[ERROR] declareWorkbenchImage(ideBundle, IDEInternalWorkbenchImages.IMG_OBJS_FIXABLE_INFO, PATH_OBJECT + "quickfix_info_obj.png", true); //$NON-NLS-1$
[ERROR] ^^^^^^^^^^^^^^^^^^^^^
[ERROR] IMG_OBJS_FIXABLE_INFO cannot be resolved or is not a field
[ERROR] /Users/claudiorosati/Projects/GitHub/cs-studio/core/utility/utility-plugins/org.csstudio.utility.product/src/org/csstudio/utility/product/ApplicationWorkbenchAdvisor.java:[176]
[ERROR] declareWorkbenchImage(ideBundle, IDEInternalWorkbenchImages.IMG_ETOOL_PROBLEMS_VIEW_INFO, PATH_EVIEW + "problems_view_info.png", true); //$NON-NLS-1$
[ERROR] ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[ERROR] IMG_ETOOL_PROBLEMS_VIEW_INFO cannot be resolved or is not a field
[ERROR] 2 problems (2 errors)
```

kasemir, 2017-05-31:
Yes, it requires Eclipse Neon as a target platform.
See #2054, where cs-studio was updated, but the products at the time had not.
Have you completed the update of the ITER product, and now all builds again?

claudio-rosati, 2017-05-31:
Sorry but I don't understand.

I'm expecting that after a merge my Jenkins pipelines and shell scripts on the community repositories and on my forked versions still work. But this is not the case. Both pipelines/scripts breaks not in my product, but in the csstudio.core module (org.csstudio.utility.product). And this happened after pulling the last changes from the master branch.

Moreover, I've not understood which changes have to be done in my product pom file. Are there documented somewhere ([#2054](https://github.com/ControlSystemStudio/cs-studio/issues/2054) does not clarify to me which files have to be changed and how)?

For me this is a serious issue because next week I'll have a summer student starting his work on porting the Jenkins pipelines from ver. 1.6.x to the new 2.x model, with the aim to share it to the community... but the current pipeline just stopped working.

boryto, 2017-05-31:
Hi Claudio, 
I think this is related to some maven caching issue. Try to delete the .m2/repositroy/p2 and .cache directories. If his doesn't help, eventually the whole .m2/repositroy can be deleted. 

The travis-CI (https://travis-ci.org/ControlSystemStudio/cs-studio) build had no problems, we also had at  local builds on different systems (Linux, Windows). 

We have build ITER and general products and there were no problems even if product pom.xml (https://github.com/ControlSystemStudio/org.csstudio.iter/blob/master/pom.xml) or (https://github.com/ControlSystemStudio/org.csstudio.product/blob/master/pom.xml) was still on mars version. 

So the product should build even if no changes are made to product pom.xml.

The changes to product pom file can taken from ITER product https://github.com/ControlSystemStudio/org.csstudio.iter/blob/master/pom.xm
In general you only have to update the 
 <eclipse-site>http://download.eclipse.org/releases/neon</eclipse-site>
 <eclipse-update-site>http://download.eclipse.org/eclipse/updates/4.6</eclipse-update-site>
to update the product to neon version. But in pom.xml other components have been updated too (orbit tools, RAP, ....). 

I hope this clears things out.

claudio-rosati, 2017-05-31:
Hello @boryto,
I've removed my local ~/.m2/repository (usually I test first on my mac with a shellscript before doing any change on the Jenkins pipelines) and the problems still exists:
```
[ERROR] Failed to execute goal org.eclipse.tycho:tycho-compiler-plugin:1.0.0:compile (default-compile) on project org.csstudio.utility.product: Compilation failure: Compilation failure:
[ERROR] /Users/claudiorosati/Projects/GitHub/cs-studio/core/utility/utility-plugins/org.csstudio.utility.product/src/org/csstudio/utility/product/ApplicationWorkbenchAdvisor.java:[156]
[ERROR] declareWorkbenchImage(ideBundle, IDEInternalWorkbenchImages.IMG_OBJS_FIXABLE_INFO, PATH_OBJECT + "quickfix_info_obj.png", true); //$NON-NLS-1$
[ERROR] ^^^^^^^^^^^^^^^^^^^^^
[ERROR] IMG_OBJS_FIXABLE_INFO cannot be resolved or is not a field
[ERROR] /Users/claudiorosati/Projects/GitHub/cs-studio/core/utility/utility-plugins/org.csstudio.utility.product/src/org/csstudio/utility/product/ApplicationWorkbenchAdvisor.java:[176]
[ERROR] declareWorkbenchImage(ideBundle, IDEInternalWorkbenchImages.IMG_ETOOL_PROBLEMS_VIEW_INFO, PATH_EVIEW + "problems_view_info.png", true); //$NON-NLS-1$
[ERROR] ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[ERROR] IMG_ETOOL_PROBLEMS_VIEW_INFO cannot be resolved or is not a field
[ERROR] 2 problems (2 errors)
```

It happens when I executhe the following:

```
(cd cs-studio/core; time mvn $MVNOPT --settings ../../ess-css-extra/maven/settings-no-tests.xml clean clean verify) | tee 3_cs-studio-core.log
```

Have you any clue?

claudio-rosati, 2017-05-31:
I've tried something different... but it fails again.

- I've removed from my composite repo files the references to 
    - cs-studio-eclipse-4.3/p2repo,
    - cs-studio-4.4/p2repo;
- I've successfully updated `pom.xml` of cs-studio-thirdparty to use neon/4.6;
- When cs-studio/core was built it failed because `javax.servlet` was not found, so I've added `<artifact><id>org.eclipse.ecf:javax.servlet:3.1.0.v201410161800</id></artifact>` to maven-osgi-repository;
- Trying again building cs-studio/core now it breaks with the following error:

```
[ERROR] Cannot resolve project dependencies:
[ERROR]   Software being installed: org.csstudio.logging.ui 3.2.2.qualifier
[ERROR]   Missing requirement: org.csstudio.logging.ui 3.2.2.qualifier requires 'bundle org.eclipse.ui.console 3.5.0' but it could not be found
[ERROR]
[ERROR] See http://wiki.eclipse.org/Tycho/Dependency_Resolution_Troubleshooting for help.
[ERROR] Cannot resolve dependencies of MavenProject: org.csstudio:org.csstudio.logging.ui:3.2.2-SNAPSHOT @ /Users/claudiorosati/Projects/GitHub/cs-studio/core/utility/utility-plugins/org.csstudio.logging.ui/pom.xml: See log for details -> [Help 1]
```

So I'm switching back to my old poms and repo files. I need help.

claudio-rosati, 2017-05-31:
P.S.
All repositories consists of the last version of the corresponding master branch.

boryto, 2017-05-31:
Hi, I would start with deleting (comment-out) all references form compositeXXXX.xml, except cs-studio-thirdparty, maven-osgi-bundles and diirt, and then try to build the core neon references in pom.xml.

claudio-rosati, 2017-05-31:
Same behavior while compiling cs-studio/core:

```
CI0011766:GitHub claudiorosati$ (cd cs-studio/core; time mvn $MVNOPT --settings ../../ess-css-extra/maven/settings-no-tests.xml clean verify) | tee 3_cs-studio-core.log
[INFO] Scanning for projects...
[INFO] Computing target platform for MavenProject: org.csstudio:org.csstudio.data:3.2.0-SNAPSHOT @ /Users/claudiorosati/Projects/GitHub/cs-studio/core/utility/utility-plugins/org.csstudio.data/pom.xml
[INFO] Adding repository file:/Users/claudiorosati/Projects/GitHub/ess-css-extra/ess_css_comp_repo
[INFO] Resolving dependencies of MavenProject: org.csstudio:org.csstudio.data:3.2.0-SNAPSHOT @ /Users/claudiorosati/Projects/GitHub/cs-studio/core/utility/utility-plugins/org.csstudio.data/pom.xml
[INFO] {osgi.os=linux, osgi.ws=gtk, org.eclipse.update.install.features=true, osgi.arch=x86_64}
[ERROR] Cannot resolve project dependencies:
[ERROR]   You requested to install 'bundle javax.servlet 0.0.0' but it could not be found
[ERROR]
[ERROR] See http://wiki.eclipse.org/Tycho/Dependency_Resolution_Troubleshooting for help.
[ERROR] Cannot resolve dependencies of MavenProject: org.csstudio:org.csstudio.data:3.2.0-SNAPSHOT @ /Users/claudiorosati/Projects/GitHub/cs-studio/core/utility/utility-plugins/org.csstudio.data/pom.xml: See log for details -> [Help 1]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MavenExecutionException

real	0m5.386s
user	0m17.540s
sys	0m0.855s
```

boryto, 2017-06-01:
Hi, I see that in your maven options you only have ess-css-settings enabled.
MVNOPT="-B -P ess-css-settings"

To get the latest eclipse bundles you should have eclipse-sites profile enabled. Try adding eclipse-sites.
We have this profiles enabled during our build: -P 'ess-css-settings,platform-default,csstudio-composite-repo-enable,eclipse-sites'

claudio-rosati, 2017-06-01:
Much better, thank you @boryto.

I still have 2 problems:

1. I had to remove `eclipse-sites` from my `MVNOPT` otherwise the following waring occurs:

```
[WARNING] The requested profile "eclipse-sites" could not be activated because it does not exist.
```

2. Display Builder doesn't build (@kasemir I need your help here). Here the error:

```
[ERROR] Cannot resolve project dependencies:
[ERROR]   Software being installed: org.csstudio.trends.databrowser3.opiwidget 4.1.0.qualifier
[ERROR]   Missing requirement: org.csstudio.utility.batik 1.1.0.201706010721 requires 'bundle org.apache.batik.bridge 1.7.0' but it could not be found
[ERROR]   Cannot satisfy dependency: org.csstudio.opibuilder.widgets 4.0.0.201706010948 depends on: bundle org.csstudio.swt.widgets 0.0.0
[ERROR]   Cannot satisfy dependency: org.csstudio.swt.widgets 2.1.4.201706010948 depends on: bundle org.csstudio.utility.batik 0.0.0
[ERROR]   Cannot satisfy dependency: org.csstudio.trends.databrowser3.opiwidget 4.1.0.qualifier depends on: bundle org.csstudio.opibuilder.widgets 1.0.5
[ERROR]
[ERROR] See http://wiki.eclipse.org/Tycho/Dependency_Resolution_Troubleshooting for help.
[ERROR] Cannot resolve dependencies of MavenProject: org.csstudio:org.csstudio.trends.databrowser3.opiwidget:4.1.0-SNAPSHOT @ /Users/claudiorosati/Projects/GitHub/org.csstudio.display.builder/databrowser3/databrowser-plugins/org.csstudio.trends.databrowser3.opiwidget/pom.xml: See log for details -> [Help 1]
```

It happens when I run the followin:

```
(cd org.csstudio.display.builder; time mvn $MVNOPT --settings ../ess-css-extra/maven/settings-no-tests.xml -Dcss-repo=file:/Users/claudiorosati/Projects/GitHub/ess-css-extra/ess_css_comp_repo clean verify) | tee 5_org.csstudio.display.builder.log
```

At this stage the repo files are the following:

```
<?xml version="1.0" encoding="UTF-8"?>
<?compositeArtifactRepository version='1.0.0'?>
<repository name="Local Composite Repository" type="org.eclipse.equinox.internal.p2.artifact.repository.CompositeArtifactRepository" version="1.0.0">
  <properties size="3">
    <property name="p2.compressed" value="true"/>
    <property name="p2.atomic.composite.loading" value="false"/>
    <!-- get new time w/ `date +%s000` -->
    <property name="p2.timestamp" value="1395756034000"/>
  </properties>
  <children size="5">
    <child location="../../maven-osgi-bundles/repository/target/repository"/>
    <child location="../../cs-studio-thirdparty/repository/target/repository"/>
    <child location="../../diirt/p2/target/repository"/>
    <child location="../../cs-studio/core/p2repo"/>
    <child location="../../cs-studio/applications/p2repo"/>

<!--
    <child location="../../org.csstudio.product/repository/target/repository"/>
    <child location="../../org.csstudio.display.builder/repository/target/repository"/>
    <child location="http://download.controlsystemstudio.org/issue/1818"/>
    <child location="../../../CS-Studio-Libraries/cs-studio-eclipse-4.3/p2repo"/>
    <child location="../../../CS-Studio-Libraries/cs-studio-4.4/p2repo"/>
 -->

  </children>
</repository>
```

and 

```
<?xml version='1.0' encoding='UTF-8'?>
<?compositeMetadataRepository version='1.0.0'?>
<repository name='Local Composite Repository' type='org.eclipse.equinox.internal.p2.metadata.repository.CompositeMetadataRepository' version='1.0.0'>
  <properties size="3">
    <property name="p2.atomic.composite.loading" value="false"/>
    <property name="p2.compressed" value="true"/>
    <!-- get new time w/ `date +%s000` -->
    <property name="p2.timestamp" value="1395756034000"/>
  </properties>
  <children size="5">
    <child location="../../maven-osgi-bundles/repository/target/repository"/>
    <child location="../../cs-studio-thirdparty/repository/target/repository"/>
    <child location="../../diirt/p2/target/repository"/>
    <child location="../../cs-studio/core/p2repo"/>
    <child location="../../cs-studio/applications/p2repo"/>

<!--
    <child location="../../org.csstudio.product/repository/target/repository"/>
    <child location="../../org.csstudio.display.builder/repository/target/repository"/>
    <child location="http://download.controlsystemstudio.org/issue/1818"/>
    <child location="../../../CS-Studio-Libraries/cs-studio-eclipse-4.3/p2repo"/>
    <child location="../../../CS-Studio-Libraries/cs-studio-4.4/p2repo"/>
 -->

  </children>
</repository>
```

claudio-rosati, 2017-06-02:
@boryto and @kasemir I really need your help in getting CS-Studio being built!

For now I skipped building `org.csstudio.display.builder` because of the missing 'bundle org.apache.batik.bridge 1.7.0'. This library is on my local p2 repo, I've tried to add a plug-in dependency inside the `org.csstudio.trends.databrowser3.opiwidget` plug-in and/or feature but without succes. So I've decided to move on waiting for some help from @kasemir.

So I'm now trying building the `org.csstudio.product` without success.

My composite repo files now contains

```xml
  <children size="6">
    <child location="../../maven-osgi-bundles/repository/target/repository"/>
    <child location="../../cs-studio-thirdparty/repository/target/repository"/>
    <child location="../../diirt/p2/target/repository"/>
    <child location="../../cs-studio/core/p2repo"/>
    <child location="../../cs-studio/applications/p2repo"/>
    <child location="http://download.controlsystemstudio.org/issue/1818"/>
  </children>
```

and the `<profile><id>platform-default</id>...</profile>` inside the `pom.xml` file was updated as follow:

```xml
<profile>
    <id>platform-default</id>
    <activation>
        <property>
            <name>!platform-version-name</name>
        </property>
    </activation>
    <properties>
        <eclipse-site>http://download.eclipse.org/releases/neon</eclipse-site>
        <eclipse-update-site>http://download.eclipse.org/eclipse/updates/4.6</eclipse-update-site>
        <rap-site>http://download.eclipse.org/rt/rap/3.2</rap-site>
        <rap-incubator-site>http://archive.eclipse.org/rt/rap/incubator/nightly/gef/20150122-1538/</rap-incubator-site>
        <platform-version>[4.4,4.6)</platform-version>
        <swtbot-site>http://download.eclipse.org/technology/swtbot/neon/dev-build/update-site</swtbot-site>
    </properties>
</profile>
<profile>
    <id>platform-mars</id>
    <activation>
        <property>
            <name>!platform-version-name</name>
            <value>mars</value>
        </property>
    </activation>
    <properties>
        <eclipse-site>http://download.eclipse.org/releases/mars</eclipse-site>
        <eclipse-update-site>http://download.eclipse.org/eclipse/updates/4.5</eclipse-update-site>
        <rap-site>http://download.eclipse.org/rt/rap/3.0</rap-site>
        <rap-incubator-site>http://archive.eclipse.org/rt/rap/incubator/nightly/gef/20150122-1538/</rap-incubator-site>
        <platform-version>[4.4,4.5)</platform-version>
        <swtbot-site>http://download.eclipse.org/technology/swtbot/mars/dev-build/update-site</swtbot-site>
    </properties>
</profile>
```

But I'm not sure it is correct.

So, what happens now is the following problem:

```
[INFO] Installing product cs-studio for environment linux/gtk/x86_64 to /Users/claudiorosati/Projects/GitHub/org.csstudio.product/repository/target/products/cs-studio/linux/gtk/x86_64/cs-studio
Installing cs-studio 4.5.1.
Installation failed.
Cannot complete the install because one or more required items could not be found.
 Software being installed: Cs-studio 4.5.1 (cs-studio 4.5.1)
 Missing requirement: toolinggtk.linux.x86_64org.eclipse.core.runtime 4.5.1 requires 'bundle org.eclipse.core.runtime 3.13.0.v20170207-1030' but it could not be found
 Cannot satisfy dependency:
  From: Cs-studio 4.5.1 (cs-studio 4.5.1)
  To: toolingcs-studio.configuration [4.5.1]
 Cannot satisfy dependency:
  From: toolingcs-studio.configuration 4.5.1
  To: toolinggtk.linux.x86_64org.eclipse.core.runtime [4.5.1]
Application failed, log file location: /Users/claudiorosati/Projects/GitHub/org.csstudio.product/workspace/.metadata/.log
```

Any idea?

claudio-rosati, 2017-06-02:
The thing I don't understand is that `org.eclipse.core.runtime-3.13.0.v20170207-1030.jar` is in my `~/.m2/repository/p2/osgi/bundle/org.eclipse.core.runtime/3.13.0.v20170207-1030` local repository.

kasemir, 2017-06-02:
I ran into the same o.e.core.runtime dependency error, which is why I started to ask about a complete updated product in #2054. Have been on travel since and had no chance to compare for example the ITER product changes with my attempt to update the SNS products.

berryma4, 2017-06-02:
Yes, we need #2054 to be resolved.  That will cause issues (the product plugin would drag in eclipse mars things). @boryto, do you have a time on this?  Or should I just update it?

claudio-rosati, 2017-06-02:
I managed to have `org.csstudio.product` successfully build on 4.6.

I still have the problem with

```
[ERROR] Cannot resolve project dependencies:
[ERROR]   Software being installed: org.csstudio.trends.databrowser3.opiwidget 4.1.0.qualifier
[ERROR]   Missing requirement: org.csstudio.utility.batik 1.1.0.201706021239 requires 'bundle org.apache.batik.bridge 1.7.0' but it could not be found
[ERROR]   Cannot satisfy dependency: org.csstudio.opibuilder.widgets 4.0.0.201706021247 depends on: bundle org.csstudio.swt.widgets 0.0.0
[ERROR]   Cannot satisfy dependency: org.csstudio.swt.widgets 2.1.4.201706021247 depends on: bundle org.csstudio.utility.batik 0.0.0
[ERROR]   Cannot satisfy dependency: org.csstudio.trends.databrowser3.opiwidget 4.1.0.qualifier depends on: bundle org.csstudio.opibuilder.widgets 1.0.5
```

in Display Builder.

boryto, 2017-06-02:
Hi, on community product we had no big issues, just checked the build and it was OK -- no errors. 

For maven build only the pom.xml sites were added.  But some more was added for eclipse in 
org.csstudio.product.ide.dependencies.feature.

I will publish the general product on monday. 

```

> [INFO] ------------------------------------------------------------------------
> [INFO] Reactor Summary:
> [INFO] 
> [INFO] product ............................................ SUCCESS [  0.102 s]
> [INFO] products-csstudio-plugins .......................... SUCCESS [  0.002 s]
> [INFO] org.csstudio.product ............................... SUCCESS [  1.586 s]
> [INFO] org.csstudio.startup.intro ......................... SUCCESS [  1.309 s]
> [INFO] products-csstudio-features ......................... SUCCESS [  0.001 s]
> [INFO] org.csstudio.product.feature ....................... SUCCESS [  0.443 s]
> [INFO] org.csstudio.product.configuration.feature ......... SUCCESS [  0.270 s]
> [INFO] repository ......................................... SUCCESS [01:12 min]
> [INFO] ------------------------------------------------------------------------
> [INFO] BUILD SUCCESS
> [INFO] ------------------------------------------------------------------------
> [INFO] Total time: 02:43 min
> [INFO] Finished at: 2017-06-02T15:28:09+02:00
> [INFO] Final Memory: 180M/650M
> [INFO] ------------------------------------------------------------------------
> 

```

claudio-rosati, 2017-06-02:
Yes, I've updated my general product pom.xml following what you've done in your and everything now is working.

The only crazy thing remain my incapacity to have the Display Builder `org.csstudio.trends.databrowser3.opiwidget` being able to see the 'bundle org.apache.batik.bridge 1.7.0' that is available in my local `.m2/repository/p2 repository`. Really frustrating.

claudio-rosati, 2017-06-02:
I managed to build als ess product (keeping Display Builder on mars).

claudio-rosati, 2017-06-02:
A last note: I run the ess product and I'm very happy to see back again the navigator icons at their full size.

First tests seems OK: BOY and Display Builder are happily running.

The only problems is the following exception, probably cause by Display Builder being on mars:

```
2017-06-02 17:33:42.027 SEVERE [Thread 28] org.csstudio.logging.PluginLogListener (logging) - Plug-in org.eclipse.egit.ui was unable to load class org.eclipse.egit.ui.internal.decorators.GitLightweightDecorator.
java.lang.ClassNotFoundException: An error occurred while automatically activating bundle org.eclipse.egit.ui (284).
	at org.eclipse.osgi.internal.hooks.EclipseLazyStarter.postFindLocalClass(EclipseLazyStarter.java:116)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.findLocalClass(ClasspathManager.java:529)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.findLocalClass(ModuleClassLoader.java:325)
	at org.eclipse.osgi.internal.loader.BundleLoader.findLocalClass(BundleLoader.java:345)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClassInternal(BundleLoader.java:423)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:372)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:364)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.loadClass(ModuleClassLoader.java:161)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	at org.eclipse.osgi.internal.framework.EquinoxBundle.loadClass(EquinoxBundle.java:564)
	at org.eclipse.core.internal.registry.osgi.RegistryStrategyOSGI.createExecutableExtension(RegistryStrategyOSGI.java:174)
	at org.eclipse.core.internal.registry.ExtensionRegistry.createExecutableExtension(ExtensionRegistry.java:905)
	at org.eclipse.core.internal.registry.ConfigurationElement.createExecutableExtension(ConfigurationElement.java:243)
	at org.eclipse.core.internal.registry.ConfigurationElementHandle.createExecutableExtension(ConfigurationElementHandle.java:55)
	at org.eclipse.ui.internal.services.WorkbenchServiceRegistry.getSourceProviders(WorkbenchServiceRegistry.java:177)
	at org.eclipse.ui.internal.services.SourceProviderService.readRegistry(SourceProviderService.java:108)
	at org.eclipse.ui.internal.Workbench$47.runWithException(Workbench.java:2447)
	at org.eclipse.ui.internal.StartupThreading$StartupRunnable.run(StartupThreading.java:32)
	at org.eclipse.swt.widgets.Synchronizer.syncExec(Synchronizer.java:233)
	at org.eclipse.ui.internal.UISynchronizer.syncExec(UISynchronizer.java:145)
	at org.eclipse.swt.widgets.Display.syncExec(Display.java:4814)
	at org.eclipse.ui.internal.StartupThreading.runWithoutExceptions(StartupThreading.java:95)
	at org.eclipse.ui.internal.Workbench.initializeDefaultServices(Workbench.java:2442)
	at org.eclipse.ui.internal.Workbench.init(Workbench.java:1678)
	at org.eclipse.ui.internal.Workbench.runUI(Workbench.java:2944)
	at org.eclipse.ui.internal.Workbench.access$9(Workbench.java:2875)
	at org.eclipse.ui.internal.Workbench$5.run(Workbench.java:684)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:610)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:148)
	at org.csstudio.utility.product.Workbench.runWorkbench(Workbench.java:103)
	at org.csstudio.startup.application.Application.startApplication(Application.java:265)
	at org.csstudio.startup.application.Application.start(Application.java:119)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:196)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:388)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:243)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:673)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:610)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1519)
Caused by: org.osgi.framework.BundleException: Exception in org.eclipse.egit.ui.Activator.start() of bundle org.eclipse.egit.ui.
	at org.eclipse.osgi.internal.framework.BundleContextImpl.startActivator(BundleContextImpl.java:795)
	at org.eclipse.osgi.internal.framework.BundleContextImpl.start(BundleContextImpl.java:724)
	at org.eclipse.osgi.internal.framework.EquinoxBundle.startWorker0(EquinoxBundle.java:932)
	at org.eclipse.osgi.internal.framework.EquinoxBundle$EquinoxModule.startWorker(EquinoxBundle.java:309)
	at org.eclipse.osgi.container.Module.doStart(Module.java:581)
	at org.eclipse.osgi.container.Module.start(Module.java:449)
	at org.eclipse.osgi.framework.util.SecureAction.start(SecureAction.java:470)
	at org.eclipse.osgi.internal.hooks.EclipseLazyStarter.postFindLocalClass(EclipseLazyStarter.java:107)
	... 44 more
Caused by: java.lang.LinkageError: loader constraint violation: when resolving method "org.slf4j.impl.StaticLoggerBinder.getLoggerFactory()Lorg/slf4j/ILoggerFactory;" the class loader (instance of org/eclipse/osgi/internal/loader/Equin [... cut ...]
	at org.eclipse.jgit.lib.Repository.<clinit>(Repository.java:110)
	at org.eclipse.egit.ui.Activator.setupRepoIndexRefresh(Activator.java:423)
	at org.eclipse.egit.ui.Activator.start(Activator.java:307)
	at org.eclipse.osgi.internal.framework.BundleContextImpl$3.run(BundleContextImpl.java:774)
	at org.eclipse.osgi.internal.framework.BundleContextImpl$3.run(BundleContextImpl.java:1)
	at java.security.AccessController.doPrivileged(Native Method)
	at org.eclipse.osgi.internal.framework.BundleContextImpl.startActivator(BundleContextImpl.java:767)
	... 51 more
```

kasemir, 2017-06-02:
Feel free to create a 'neon' branch for the display builder, so that you can compile with 100% eclipse neon.

kasemir, 2017-06-02:
I've created a 'neon' branch for the display builder. I'm stuck on a missing batik dependency, in my case org.apache.batik.css to build this branch successfully.

claudio-rosati, 2017-06-03:
Similar to what happens to me. I've tried to add the missing project into the manifest, both as a requirement and an import, without success. I've tried to add a dependency in the feature. Idem. I've tried both...

Eclipse + Maven: the magic of!

kasemir, 2017-06-04:
Some new error pops up whenever I try to fix one thing.
To understand and work around the missing ‘batik’ dependencies, I switched to a  local target platform, no http:// links at all to eclipse, orbit, ..., so I should know 100% what’s in there.

I populated that with the Eclipse Neon.3 IDE etc. core, applications build OK (and faster).
SNS products now all fail with missing dependency on org.eclipse.osgi 3.12.0.v20170118-2208.

That is a curious version number. The Neon.3 target platform contains org.eclipse.osgi 3.11.3.v20170209-1843. That’s a lower version number (3.11 instead of 3.12) but later build date (Feb 2017 instead of Jan 2017).
The 3.12 version is for Eclipse Oxygen (4.7), not Neon (4.6).
Somehow a dependency on an older build of the not-yet-released Oxygen version sneaks in.

I find no feature, plugin or product that specifically requires org.eclipse.osgi 3.12.
Nothing in the maven output mentions org.eclipse.osgi 3.12 up the error message, i.e. no prior “Downloading..” message, and in fact it should not download anything beyond maven plugins because there are http://.. links to P2 repositories.
When I delete the ~/.m2/repository to start over, maven does of course download the java and tycho plugins to learn how to do things, but there are no downloads for the target platform in the maven output. 
Still, the .m2/repository sure contains the org.eclipse.osgi 3.12.0.v20170118-2208 jar, so it is silently obtained from somewhere.

Even if I then add that org.eclipse.osgi 3.12 jar to my target platform, there’s still the “not found” error. Eventually, I found that the file repository/target/p2content.xml that’s created by the tycho/maven build contains this:
````
  <unit id='toolingcocoa.macosx.x86_64org.eclipse.osgi'
          version='4.2.0.201706040134' singleton='false'>
    <hostRequirements size='2'>
      <required namespace='osgi.bundle' name='org.eclipse.osgi' range='3.12.0.v20170118-2208'/>
````
“tooling...” must be an auto-created unit that tycho uses to assemble the product, and in there it seems that tycho decides to require the pre-release version of org.eclipse.osgi for Eclipse Oxygen.

kasemir, 2017-06-04:
Going back to building 'master' with the pom.xml repository entries for eclipse, eclipse-updates, orbit enabled, it also downloads org.eclipse.osgi_3.12.0.v20170118-2208.jar:

````
[INFO] Resolving dependencies of MavenProject: org.csstudio:org.csstudio.data:3.2.0-SNAPSHOT @ /home/training/git/cs-studio/core/utility/utility-plugins/org.csstudio.data/pom.xml
[INFO] Fetching com.google.guava_21.0.0.v20170206-1425.jar.pack.gz from http://download.eclipse.org/tools/orbit/downloads/drops2/R20170307180635/repository/plugins/ (676.25kB)
[INFO] Fetching org.eclipse.osgi_3.12.0.v20170118-2208.jar.pack.gz from http://download.eclipse.org/tools/orbit/downloads/drops2/R20170307180635/repository/plugins/ (433.63kB)
````

This unpredictability is maddening: We try to select Eclipse Neon as part of the target platform, but since our target platform is basically just "Here, use any of these remote sites and see what you find", in this case Orbit happens to have some version of org.eclipse.osgi that's in fact older, but has a newer looking version number.

kasemir, 2017-06-04:
Progress?

The `batik` dependencies are solved by adding Orbit to the repositories.
For some reason the SNS product pom originally didn't need orbit, and I assume the ESS product was copied from there and thus also didn't include orbit.

The Eclipse Orbit downloads page suggests what's shown in https://github.com/ControlSystemStudio/cs-studio/blob/master/core/pom.xml#L24, 
```
http://download.eclipse.org/tools/orbit/R-builds/R20170307180635/repository
```
for Neon.3, i.e. the most recent version of Eclipse Neon 4.6.3.
But now Orbit results in the strange dependency on  org.eclipse.osgi_3.12.0.v20170118-2208 in my build or org.eclipse.core.runtime 3.13.0.v20170207-1030 in Claudio's example that he copied above.

When I switch `<orbit-site>` to the first Neon release, I don't get the strange org.eclipse.osgi version dependency:
```
http://download.eclipse.org/tools/orbit/downloads/drops/R20160520211859/repository
```

kasemir, 2017-06-04:
@claudio-rosati, regarding your reported error
````
Software being installed: Cs-studio 4.5.1 (cs-studio 4.5.1)
 Missing requirement: toolinggtk.linux.x86_64org.eclipse.core.runtime 4.5.1 requires 'bundle org.eclipse.core.runtime 3.13.0.v20170207-1030' but it could not be found
````

Did this only happen when building on Mac OS X?

I see a similar error
````
Software being installed: Css 4.2.0.201706041653 (org.csstudio.basic.epics.product.product 4.2.0.201706041653)
 Missing requirement: toolinggtk.linux.x86org.eclipse.osgi 4.2.0.201706041653 requires 'bundle org.eclipse.osgi 3.12.0.v20170118-2208' but it could not be found
````

This happens on Mac OS X, when the build tries to assemble the product for Linux.
On Linux, it builds for all platforms without that error. Again the version number of the core.runtime or eclipse.osgi is not the one that goes with Eclipse Neon.

For now my conclusion is that something in tycho for building on Mac is depending on a pre-release of Eclipse Oxygen.

claudio-rosati, 2017-06-04:
I've not tried on Linux yet.

kasemir, 2017-06-04:
I've just updated the 'master' for the display builder to include the neon and orbit repository links, and to fix some new compiler errors.

On Linux, I can now build the 'basic epics' product for all platforms, and the result for Mac OS seems to run OK on Mac OS, with BOY and Display Builder.
Will push the updated SNS product 'master' by end of day so you can compare.

I'm still horrified by the randomness. We need to update 'mars' to 'neon' in all the top-level pom.xml, and orbit drags in versions that are not a match for 'neon'. Point is we really need a "target platform" that is 100% reproducible.

claudio-rosati, 2017-06-07:
A last note: I run the ess product and I'm very happy to see back again the navigator icons at their full size.

First tests seems OK: BOY and Display Builder are happily running.

I still have a lot of problems having the product built succesfully... it happens randomly. Most of the time not. I'm still investigating.

Generally speaking, I think moving to Neon was not a good choice and, mostly, it was performed badly.

claudio-rosati, 2017-06-07:
I've managed to build all the ess product for Neon without unit test.
When unit tests are performed too, I have a problem interrupting the buld: #2218

claudio-rosati, 2017-06-09:
A part the #2218 issue, my Jenkins build is back to normal.
I close this issue.
Thank you all for the help.

closed 2017-06-09 by claudio-rosati