ControlSystemStudio/cs-studio #2213: Data browser cannot open savedfile

MjBordalo, 2017-05-31:
Hi i am having a problem with BOY.
After i save a databrowser file and try to reopen it the working window gets grey (no databrowser layout).
Then if i close that windows i get this error:
An error has occurred. See error log for more details.
java.lang.NullPointerException

Help apreciated
Thanks in advance
MiguelB

MjBordalo, 2017-05-31:
solved by unchecking:
`Use default  archives, ignoring *.plt file?`
under Edit >Preferences> CSS Applications>Trends>Data Browser

kasemir, 2017-05-31:
Re-opening issue since I'll need to look into fixing the NullPointerException.
Good that you found a way to work around the error.

kasemir, 2017-06-04:
Under Edit >Preferences> CSS Applications>Trends>Data Browser, do you have any default data sources configured, or is the list empty?

MjBordalo, 2017-06-05:
Indeed. I have the same configurations as in Archive Data Servers URLs.
Was not supposed?

kasemir, 2017-06-05:
Can you add a screenshot of your Data Browser settings, i.e. the ones that result in the null pointer error when you _check_ `Use default archives, ignoring *.plt file`?

MjBordalo, 2017-06-05:
i believe it was like this.

[image]