ControlSystemStudio/cs-studio #2217: Full ISO time-stamp formatting

boryto, 2017-06-06:
ITER has a requirement to use full ISO formatting (with letter T and zone).
To satisfy the requirement we propose an addition to TimestampFormats class. 

Additional setting will be defined: org.csstudio.java/use_full_iso_timestamp_formatting. 
When this is true, the patterns change to:
FULL_FORMAT =  "yyyy-MM-dd'T'HH:mm:ss.nnnnnnnnnX" -- ex. 2017-06-06T11:38:08.057000000Z
MILLI_FORMAT = "yyyy-MM-dd'T'HH:mm:ss.SSSX" -- ex. 2017-06-06T11:38:08.057Z
SECONDS_FORMAT = "yyyy-MM-dd'T'HH:mm:ssX" --ex. 2017-06-06T11:38:08Z

The preference will be false by default, so no change is expected for existing products.

boryto, 2017-06-07:
The org.csstudio.java.time.TimestampFormats has no Eclipse dependencies. If we want to read the preferences from the .ini file, the dependency on org.eclipse.core.runtime.preferences has to be added to the class.

Will this be OK?

css-iter, 2017-06-08:
#2219

boryto, 2017-06-08:
Created a pull request #2219

boryto, 2017-06-28:
Added formatting to Logbook. PR: #2223

css-iter, 2017-07-07:
Added to Databrowser, JMS monitor and Alarm displays: #2232

berryma4, 2017-09-05:
close?

boryto, 2017-09-06:
Yes, If there will be any new additions, we can create new one or reopen this one.

closed 2017-09-06 by boryto