ControlSystemStudio/cs-studio #2218: Unable to perform build with tests of cs-studio/applications

claudio-rosati, 2017-06-07:
Hello,

I'm still unable to perform a full build with tests. I'm not sure if this is related to #2210, but it worked in the past, so...

The failing test is `org.csstudio.swt.rtplot.test`, with the following message:

```
Running org.csstudio.swt.rtplot.util.UpdateThrottleTest
-> Trigger..
<- Update!
-> Trigger..
-> Trigger..
-> Trigger..
<- Update!
<- Update!
Tests run: 1, Failures: 1, Errors: 0, Skipped: 0, Time elapsed: 5.018 sec <<< FAILURE! - in org.csstudio.swt.rtplot.util.UpdateThrottleTest
testUpdateThrottle(org.csstudio.swt.rtplot.util.UpdateThrottleTest)  Time elapsed: 5.015 sec  <<< FAILURE!
java.lang.AssertionError:
Expected: <2>
     but: was <3>
	at org.hamcrest.MatcherAssert.assertThat(MatcherAssert.java:20)
	at org.junit.Assert.assertThat(Assert.java:956)
	at org.junit.Assert.assertThat(Assert.java:923)
	at org.csstudio.swt.rtplot.util.UpdateThrottleTest.testUpdateThrottle(UpdateThrottleTest.java:65)

Results :

Failed tests:
  UpdateThrottleTest.testUpdateThrottle:65
Expected: <2>
     but: was <3>
```

boryto, 2017-06-07:
Hi, I check the test locally -- from the development environment in Neon, and it passed.  So probably not in direct connection to Neon.

claudio-rosati, 2017-06-08:
Thank you @boryto.
Is there someone knowing what the test is performing, able to help me?

claudio-rosati, 2017-06-09:
@berryma4 @kasemir @shroffk @willrogers 

Any idea about this problem? 
I've commented out lines 62-75 of `org.csstudio.swt.rtplot.util.UpdateThrottleTest` and verified that no other problems exist in building `cs-studio/applications`, so this one is the only critical one.

kasemir, 2017-07-03:
The test depends on timing.
A flurry of 3 update triggers should be collapsed into one combined update.
On the computer where you run the test, the flurry of triggers is executed with a delay, so one of them is passed on as another individual update.

The test is already somewhat lenient. For things that should happen in ~1 second, it waits 4 seconds to allow 3 extra seconds. Still, all timing-based tests can run into occasional errors. Especially printouts can skew the timing.
What you could try is change this
````
// Follow-up triggers are delayed...
System.out.println("-> Trigger..");
throttle.trigger();
System.out.println("-> Trigger..");
throttle.trigger();
System.out.println("-> Trigger..");
throttle.trigger();
````
into this:
````
// Follow-up triggers are delayed...
System.out.println("-> Trigger three times..");
throttle.trigger();
throttle.trigger();
throttle.trigger();
````

claudio-rosati, 2017-07-05:
@kasemir 
With the change you suggested the test performed well.
So which version of the test do you want to keep in GitHub (i.e. Have I to do a pull-request)?

kasemir, 2017-07-05:
Since you got the test to pass, can you create a pull request?

berryma4, 2017-09-05:
close?