ControlSystemStudio/cs-studio #2222: Pv manager new diirt vImage

boryto, 2017-06-23:
The new dirt vImage getData method changed. Retun type is now ListNumber (before byte[]).

Org.csstudio.utility.pvmanager.ui has to be changed to reflect the changes. 

`ImageData imageData = new ImageData(vImage.getWidth(), vImage.getHeight(), 24, new PaletteData(0xFF, 0xFF00, 0xFF0000), vImage.getWidth()*3, vImage.getData());`

```
[ERROR] ImageData imageData = new ImageData(vImage.getWidth(), vImage.getHeight(), 24, new PaletteData(0xFF, 0xFF00, 0xFF0000), vImage.getWidth()*3, vImage.getData());
[ERROR] ^^^^^^^
[ERROR] The method getData() from the type VImage refers to the missing type ListNumber
```

boryto, 2017-06-29:
A quick fix would be changing:

```
 public static Image toImage(GC gc, VImage vImage) {
    ImageData imageData = new ImageData(vImage.getWidth(), vImage.getHeight(), 24, new PaletteData(0xFF, 0xFF00, 0xFF0000), vImage.getWidth()*3, vImage.getData());
     Image image = new Image(gc.getDevice(), imageData);
    return image;
```

to:

```
  public static Image toImage(GC gc, VImage vImage) {
        ListNumber data = vImage.getData();
        byte imageByteData[] = new byte[data.size()];
        for (int i = 0; i < data.size(); i++) {
            imageByteData[i] = data.getByte(i);
        }
        ImageData imageData = new ImageData(vImage.getWidth(), vImage.getHeight(), 24, new PaletteData(0xFF, 0xFF00, 0xFF0000), vImage.getWidth() * 3, imageByteData);
        Image image = new Image(gc.getDevice(), imageData);
        return image;
    }
```
but I am not sure this is the proper way. 
Can anyone that designed the DIIRT vImage change reflect on this?

kasemir, 2017-06-29:
Related: #2201 updates some of core to use the new vImage API

boryto, 2017-07-11:
PR: #2231

boryto, 2017-07-14:
Fixed with #2236

closed 2017-07-14 by boryto