ControlSystemStudio/cs-studio #2228: Converting EDL files

claudio-rosati, 2017-07-04:
Hello,

I've problems converting the attached EDL files.

When I right-clik on them and select **Convert to OPI file** action I have the following message:

```
2017-07-04 12:00:59 ERROR: Converting error in file L/CSS/spectrum.edl
SPECIFIC_PARSING_ERROR exception: Error when parsing annotated field.
null
```

Have I missed some plug-in dependencies? Do you think would be better to have some more information in the error message (at least the line number of the EDL file where the parser stopped)?

[EDL-files.zip](https://github.com/ControlSystemStudio/cs-studio/files/1122009/EDL-files.zip)

claudio-rosati, 2017-07-13:
@kasemir @shroffk @berryma4 @willrogers 
Someone has an idea and can help me?

willrogers, 2017-07-13:
Hi Claudio,

We have done a lot of EDM conversion.  I can have a look.

One thing you are missing is the EDM `colors.list` file - it is like the boy `color.def` file.  I used our one but I don't think it included the right colors.

Anyway, have a look at the attached files.

[claudio.zip](https://github.com/ControlSystemStudio/cs-studio/files/1145595/claudio.zip)

Cheers,
Will

willrogers, 2017-07-13:
When I used the same method as you I got the same error as you, which did not tell me what the problem was.  When I ran the converter directly from Eclipse, it gave me the more descriptive error:

```
Jul 13, 2017 4:00:18 PM org.csstudio.opibuilder.converter.EdmConverter validateColorsFile
SEVERE: File colors.list does not exist.
Error: File colors.list does not exist.
```

Specifying the `colors.list` file then allowed me to do the conversion.

kasemir, 2017-07-13:
The colors.list file used by EDM as well as similar file then created for the *.opi needs to be entered into the preferences under CSS Applications, Display, EDM2OPI Converter.
I think the actual generated output OPI color file can then be deleted, it would only be useful if you planned to continue to use named colors of the EDM files in newly created *.opi files.

claudio-rosati, 2017-07-14:
@willrogers @kasemir 
Thank you for your help.

Now I've the following issue:

- the converter plugin has the `colors.list` file in its help folder. Specifying this file as `platform:/plugin/org.csstudio.opibuilder.converter/html/sample_edl_files/colors.list` doesn't work;
- copying the file in my configuration folder and specifying it as `platform:/config/edm-colors.list` doesn't work either.

Anyway, I thing that the converter in case such file is not specified or doesn't exist should write a better error message.

kasemir, 2017-07-14:
The colors.list file should be the one that your EDM installation uses, i.e. some file in the file system.
Not something in a plugin.jar. If you don't have the correct colors.list, you can try the example colors.list from the converter, but the converter code might not be able to read it from within a plugin, so place is somewhere in the file system and then provide /the/absolute/path/to/that/colors.list.

willrogers, 2017-07-14:
I absolutely agree about the error message, but right now I don't have time to dive in and try to fix it.

The same applies for the location of the configuration files. We could add code to interpret the `platform:` URIs (and then we could put it in a common library for any plugin to use), but it would take some time!

claudio-rosati, 2017-07-17:
@kasemir 
I understand and agree with you the color file must come with the OPI to be converted.
Thank you

@willrogers 
Thank you.

mdavidsaver, 2019-01-30:
I encountered this issue as well, but found a workaround.  EDM itself can reprocess .edl files to embed RGB value instead of color list index.  The following assumes that the .edl files, as well as their matching colors.list and fonts.list are in the current directory.

```
EDMCOLORMODE=rgb EDMFILES=$PWD EDMDATAFILES=$PWD edm -convert *.edl
```

This results in changes likes like `bgColor index 3` -> `bgColor rgb 56026 56026 56026`.
Not a perfect solution as it loses the color names, but in my case this isn't important.