ControlSystemStudio/cs-studio #2238: Unable to build cs-studio

claudio-rosati, 2017-07-17:
Hello,

after merging the last changes, I'm unable to build CS-Studio.

Here the error building `org.csstudio.utility.pvmanager.ui`:

```
[ERROR] Failed to execute goal org.eclipse.tycho:tycho-compiler-plugin:1.0.0:compile (default-compile) on project org.csstudio.utility.pvmanager.ui: Compilation failure: Compilation failure:
[ERROR] /Users/claudiorosati/Projects/GitHub/cs-studio/core/utility/utility-plugins/org.csstudio.utility.pvmanager.ui/src/org/csstudio/utility/pvmanager/ui/SWTUtil.java:[54]
[ERROR] ListNumber data = vImage.getData();
[ERROR] ^^^^^^^^^^^^^^^^
[ERROR] Type mismatch: cannot convert from byte[] to ListNumber
```

Any idea on why this occurred?

Moreover, I'm also unable to build cs-studio-thirdparty because of 

```
[ERROR] Internal error: java.lang.RuntimeException: Failed to load p2 repository with ID 'eclipse' from location http://download.eclipse.org/releases/neon/: Unable to read repository at http://download.eclipse.org/releases/neon. Unable to read repository at http://download.eclipse.org/releases/neon/201703141400. Input is not in the XZ format -> [Help 1]
00:05:18.060 org.apache.maven.InternalErrorException: Internal error: java.lang.RuntimeException: Failed to load p2 repository with ID 'eclipse' from location http://download.eclipse.org/releases/neon/
```

Have you noticed the same problem?

willrogers, 2017-07-17:
I also have problems building at the moment.

kasemir, 2017-07-17:
You need the updated diirt library, which is the master of diirt/diirt, not ControlSystemStudio/diirt.

But then you'll find that for the updated diirt, you also need #2201, which updates the vtype.pv to the new diirt VImage.

claudio-rosati, 2017-07-18:
Sorry but I don't understand:

- Has `ControlSystemStudio/diirt` to be removed from the build system?
- How `diirt/diirt` has to be added? From past discussions with @shroffk, I understood that the new DIIRT library should have been included directly from Maven central (in `maven-osgi-bundles` project, thus relieving CS-Studio of the burden of building it, and speeding up the CS-Studio build itself). Is that the case? If yes, why is it not yet in `maven-osgi-bundles` project? If not, what are the steps to be followed to have again a warking build (can the online [Control System Studio Guide](http://cs-studio.sourceforge.net/docbook/pt01.html) be updated)?

willrogers, 2017-07-18:
Like Claudio, I'm not sure how to make this change.  For example, `org.csstudio.product` fails presumably because the Diirt changes have not yet been ironed out:

```
...
[INFO] ------------------------------------------------------------------------
[INFO] Building repository 4.5.1-SNAPSHOT
[INFO] ------------------------------------------------------------------------
[INFO] 
[INFO] --- maven-clean-plugin:2.5:clean (default-clean) @ repository ---
[INFO] Deleting /scratch/hgs15624/css/org.csstudio.product/repository/target
[INFO] 
[INFO] --- tycho-packaging-plugin:1.0.0:build-qualifier-aggregator (default-build-qualifier-aggregator) @ repository ---
[INFO] The project's OSGi version is 4.5.1.201707141211
[INFO] 
[INFO] --- maven-clean-plugin:2.5:clean (default-clean-1) @ repository ---
[INFO] 
[INFO] --- maven-resources-plugin:2.4.3:resources (default-resources) @ repository ---
[INFO] Using 'UTF-8' encoding to copy filtered resources.
[INFO] skip non existing resourceDirectory /scratch/hgs15624/css/org.csstudio.product/repository/src/main/resources
[INFO] 
[INFO] --- maven-antrun-plugin:1.7:run (copy-artifacts-content-data-to-central-repository) @ repository ---
[INFO] Executing tasks

main:
    [mkdir] Created dir: /scratch/hgs15624/css/org.csstudio.product/repository/target/centralrepository
      [get] Getting: http://download.controlsystemstudio.org/product/4.5/artifacts.jar
      [get] To: /scratch/hgs15624/css/org.csstudio.product/repository/target/centralrepository/artifacts.jar
      [get] Getting: http://download.controlsystemstudio.org/product/4.5/content.jar
      [get] To: /scratch/hgs15624/css/org.csstudio.product/repository/target/centralrepository/content.jar
[INFO] Executed tasks
[INFO] 
[INFO] --- target-platform-configuration:1.0.0:target-platform (default-target-platform) @ repository ---
[INFO] 
[INFO] --- tycho-p2-publisher-plugin:1.0.0:publish-products (default-publish-products) @ repository ---
[IconExe] Error - 7 original icon(s) not replaced in /tmp/p2.brandingIron2535023116198304339/launcher.exe:
	- 48x48, 8 bits, 3752 byte(s) @ 18960
	- 32x32, 8 bits, 2216 byte(s) @ 22712
	- 16x16, 8 bits, 1384 byte(s) @ 24928
	- 48x48, 32 bits, 9640 byte(s) @ 26312
	- 32x32, 32 bits, 4264 byte(s) @ 35952
	- 16x16, 32 bits, 1128 byte(s) @ 40216
	- 256x256, 32 bits, 270376 byte(s) @ 41344
[IconExe] For more details, see https://git.eclipse.org/c/equinox/rt.equinox.p2.git/tree/bundles/org.eclipse.equinox.p2.publisher.eclipse/src/org/eclipse/pde/internal/swt/tools/IconExe.java
[INFO] 
[INFO] --- tycho-p2-publisher-plugin:1.0.0:publish-categories (default-publish-categories) @ repository ---
[INFO] 
[INFO] --- tycho-p2-publisher-plugin:1.0.0:attach-artifacts (default-attach-artifacts) @ repository ---
[INFO] 
[INFO] --- tycho-p2-repository-plugin:1.0.0:assemble-repository (default-assemble-repository) @ repository ---
[WARNING] Mirror tool: Problems resolving provisioning plan.: [Unable to satisfy dependency from cs-studio 4.5.1 to org.csstudio.core.diirt.feature.feature.group [0.0.1.201707141204].]
[WARNING] More information on the preceding warning(s) can be found here:
[WARNING] - http://wiki.eclipse.org/Tycho_Messages_Explained#Mirror_tool
[WARNING] Mirror tool: Problems resolving provisioning plan.: [Unable to satisfy dependency from cs-studio 4.5.1 to org.csstudio.core.diirt.feature.feature.group [0.0.1.201707141204].]
[WARNING] More information on the preceding warning(s) can be found here:
[WARNING] - http://wiki.eclipse.org/Tycho_Messages_Explained#Mirror_tool
[WARNING] Mirror tool: Problems resolving provisioning plan.: [Unable to satisfy dependency from cs-studio 4.5.1 to org.csstudio.core.diirt.feature.feature.group [0.0.1.201707141204].]
[WARNING] More information on the preceding warning(s) can be found here:
[WARNING] - http://wiki.eclipse.org/Tycho_Messages_Explained#Mirror_tool
[INFO] 
[INFO] --- tycho-p2-repository-plugin:1.0.0:archive-repository (default-archive-repository) @ repository ---
[INFO] Building zip: /scratch/hgs15624/css/org.csstudio.product/repository/target/repository-4.5.1-SNAPSHOT.zip
[INFO] 
[INFO] --- tycho-p2-director-plugin:1.0.0:materialize-products (materialize-products) @ repository ---
[INFO] Installing product cs-studio for environment linux/gtk/x86_64 to /scratch/hgs15624/css/org.csstudio.product/repository/target/products/cs-studio/linux/gtk/x86_64/cs-studio
Installing cs-studio 4.5.1.
Installation failed.
Cannot complete the install because one or more required items could not be found.
 Software being installed: Cs-studio 4.5.1 (cs-studio 4.5.1)
 Missing requirement: Cs-studio 4.5.1 (cs-studio 4.5.1) requires 'org.csstudio.core.diirt.feature.feature.group [0.0.1.201707141204]' but it could not be found
Application failed, log file location: /scratch/hgs15624/css/org.csstudio.product/workspace/.metadata/.log
[... 5 lines cut ...]
[INFO] org.csstudio.product ............................... SUCCESS [  0.525 s]
[INFO] org.csstudio.startup.intro ......................... SUCCESS [  0.423 s]
[INFO] products-csstudio-features ......................... SUCCESS [  0.001 s]
[INFO] org.csstudio.product.feature ....................... SUCCESS [  0.217 s]
[INFO] org.csstudio.product.configuration.feature ......... SUCCESS [  0.074 s]
[INFO] repository ......................................... FAILURE [ 18.776 s]
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[... cut ...]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException
[ERROR] 
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <goals> -rf :repository
hgs15624@pc0072 /s/h/c/org.csstudio.product> 
```

shroffk, 2017-07-18:
1. Currently I am working on getting the process in place to stage and release to maven central since I found out that the epics maven repository on Sourceforge has been  abandoned.

2. In the meanwhile I am releasing the diirt p2 repositories to the same s3 location as before, the only change is that the release is being done from diirt/diirt instead of controlSystemStudio/diirt
[image]

3. None of the product poms have any direct reference to a diirt release numbers. I had checked this while releasing the 3.1.6. The product requires, the o.s.core.diirt.feature...which is supposedly available on the 4.5 composite core repo

[image]

claudio-rosati, 2017-07-18:
Can we definitively switch to the maven version (through the `maven-osgi-bundles`) instead of having this intermediate step? How long requires having DIIRT on Maven?

I really don't want to spend time trying to have my local and Jenkins building working now, and change it again in few days for the better and cleanest version.

shroffk, 2017-07-18:
@claudio-rosati this intermediate step requires no change to the build, other than the update of the diirt.version number...if the build if failing now then we might see similar issues in the future when we update other pieces.

claudio-rosati, 2017-07-18:
Sorry @shroffk but I still don't understand.

Currently my build script (and Jenkins pipeline) are based on the instructions in [Control System Studio Guide](http://cs-studio.sourceforge.net/docbook/pt01.html): I have a forked copy of `ControlSystemStudio/diirt` that I build myself by means of the following command:

```
echo ""
echo "===="
echo "==== BUILDING diirt"
echo "===="
(cd diirt; time mvn $MVNOPT --settings ../ess-css-extra/maven/settings.xml clean verify) | tee 2_diirt.log
``` 

So what now? Have I to fork `diirt/diirt` and build it in the same manner?

shroffk, 2017-07-18:
https://github.com/ControlSystemStudio/org.csstudio.docbook/pull/12

shroffk, 2017-07-18:
@willrogers I updated the product pom.xml to include the diirt repo (which wasn't there before)
https://github.com/ControlSystemStudio/org.csstudio.product/pull/28

I also updated the pom to follow the checkstyle

claudio-rosati, 2017-07-19:
Before going to update the pom file I have another problem: now cs-studio core and applications build successfully, not display builder ( @kasemir I need your help here ):

```
[ERROR] Failed to execute goal org.eclipse.tycho:tycho-compiler-plugin:1.0.0:compile (default-compile) on project org.csstudio.display.builder.representation.javafx: Compilation failure: Compilation failure:
[ERROR] /Users/claudiorosati/Projects/GitHub/org.csstudio.display.builder/org.csstudio.display.builder.representation.javafx/src/org/csstudio/display/builder/representation/javafx/widgets/plots/ImageRepresentation.java:[297]
[ERROR] image_plot.setValue(image.getWidth(), image.getHeight(), new ArrayByte(image.getData(), true),
[ERROR] ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[ERROR] The constructor ArrayByte(ListNumber, boolean) is undefined
[ERROR] 1 problem (1 error)
```

What is still missing?

berryma4, 2017-09-05:
close?

claudio-rosati, 2017-10-11:
@berryma4 For me the issue can be closed. @willrogers can you close the issue?

closed 2017-10-11 by willrogers