ControlSystemStudio/cs-studio #2243: New feature for tabbed container wigdet

charlesHenryPATARD, 2017-07-21:
Hello,
We would like to know if it's possible to trigg actions when user change tab in tabbed conatainer widget ? By clicking on tab title.
Currently, actions are trigged by clicking inside the tab.

kasemir, 2017-07-21:
Well, you can peek into the source code of the widget, and then use a script like this one that writes the currently selected tab to a local variable.

Then you could do something else by reacting to changes in that local variable. Meaning: You could trigger other scripts.

But as always with scripts, the underlying API can change. In fact will change, the Display Builder has a different widget model and thus script API.
So then you'd have to update your script, for example by adding
````
print(widget.__class__.__name__)
````
to check what the exact widget class is that you receive in the script, then check the source code how it may allow you to subscribe to changes in the selected tab.

The essential script of the attached zip:
The Tab widget has a script that's once triggered by `loc://$(DID)_init(1)`, and it also receives a PV `loc://active_tab(0)` to which it writes the newly selected tab from a listener to tab selections:
````
from org.csstudio.swt.widgets.figures.TabFigure import ITabListener

class MyTabListener(ITabListener):
    def activeTabIndexChanged(self, old, new):
        print("Selected tab %d" % new)
        pvs[1].setValue(new) 
            
widget.getFigure().addTabListener(MyTabListener())
````

[tab.zip](https://github.com/ControlSystemStudio/cs-studio/files/1165519/tab.zip)

charlesHenryPATARD, 2017-07-24:
Hello,
Thanks Kay we will try.