ControlSystemStudio/cs-studio #2244: Graphical bug on linking container widget

charlesHenryPATARD, 2017-07-21:
Hello,
We think that there is a graphical bug in linking container widget. We've created an opi with 3 linking container widgets linked to the same opi file.
When we move the mouse over the second linking container widget, the first one is activated. And when we click into the second one the check box is checked.
I've tested with the last version of CS-Studio on Linux. 

I prepared opi files for you : 
- Execute ICV196-JN.opi 
- Intialize PVs : loc://icv196_co-j1:Write16,  loc://icv196_co-j2:Write48 and  loc://icv196_co-j3:Write80 with text input
- Move the mouse into the second linking container (loc://icv196_co-j2) just under BB, BA, B9 text
And you will see check box selected on loc://icv196_co-j1 linking container.

You download files here : https://u.ganil-spiral2.eu/cloud/index.php/s/mpwoVyRmLpvSBTj.

kasemir, 2017-07-21:
I assume you mean "Execute ICV196_ReadWrite.opi", because "ICV196-JN.opi" is the one that's linked?

It works for me.
When I slowly move the mouse around in the middle embedded display, I can see that some of the checkboxes in the top embedded displays appear to become 'armed', but when I actually press a button, it's always the correct one.
That strange activation of buttons has been described in fixed in #1818. It's a bug in SWT, it confuses the mouse coordinates relative to the container with the coors relative to the origin of the display, so we added code that checks on click if this is really the button that we wanted to click.

Check if you have the latest source code from 'master'.

charlesHenryPATARD, 2017-07-24:
Hello Kay,
That's it. If you want to test you have to execute ICV196_ReadWrite.opi. We've tested with Linux OS. May be it's depend on which OS you use ? 
In fact, this bug seems to be the same as  #1818. So I will test with sources from master. I close this issue.

closed 2017-07-24 by charlesHenryPATARD