ControlSystemStudio/cs-studio #2249: The CSS Core ➜ Process Variables ➜ Loc DataSource needs to be updated

claudio-rosati, 2017-08-02:
The last table on the page is outdated, not showing the `loc://pvname<VEnum>` type.

Probably the whole page has to be updated to the currently supported `log://` data souurce.

willrogers, 2017-08-02:
The local PV part is my job.  Right now I can't build CSS due to a conflict between `cs-studio/core` and `diirt` - is that true for everyone?

claudio-rosati, 2017-08-02:
I had the problem, now its ok. What you have to do is using the diirt/diirt GitHup projectm not the ControlSystemStudio/diirt one.

claudio-rosati, 2017-08-02:
Another question about the local PV, it seems to me that VBoolean is not supported. Is ti correct? If yes, why?

willrogers, 2017-08-02:
Historically, I think that VInt wasn't added because VDouble would suffice, and perhaps the same is true for VBoolean.  Do you have a specific need?

kasemir, 2017-08-02:
In practice, there really is no VBoolean PV.
Boolean PVs, for example from bi, bo records, are VEnum with two states.

claudio-rosati, 2017-08-03:
No, not a specific need. But when in the code I see the VBoolean class, I'm expecting to be able to use it also in the local PV because, semantically, a boolean is not an int, nor an enum.

closed 2017-10-05 by willrogers