ControlSystemStudio/cs-studio #2263: Archive Engine does not Archive Arrays

DominicOram, 2017-08-21:
At ISIS we recently upgraded our ArchiveEngine from version 3.1. 

It appears that in the latest version when a waveform PV is set to be archived the data is not saved into the database. A new entry is created in the sample table when the PV changes and the first element of the array is put into the float_val however the array_val remains null. This has been tested using float and char arrays, neither of which work.

kasemir, 2017-08-22:
Have you checked if the engine is configured to use the `array_val` blob or not?
By default it's disabled, and then the values end in the array_val table vs. the samples.array_val blob:
```
org.csstudio.archive.rdb/use_array_blob=false
```

kasemir, 2017-08-22:
I just tried with the latest from 'master', using the 'Demo' configuration which includes `sim://noiseWaveform(0,10,100,10)`. By default, it writes to the `array_val` table. When I change the configuration to ` org.csstudio.archive.rdb/use_array_blob=true`, it writes to the `sample.array_val` blob.

DominicOram, 2017-08-22:
Oops! Thank you!

closed 2017-08-22 by DominicOram