ControlSystemStudio/cs-studio #2267: Release 4.5.x (master)

berryma4, 2017-08-28:
Proposing to freeze the master branch to 4.5.x.
- [] Close/merge pull request #2205

Site Representatives, are there any other outstanding issues that need to be addressed before release of 4.5.x?

- [x] @berryma4
- [ ]  @kasemir
- [ ] @shroffk , 
- [ ] @utzeln, 
- [ ] @mattclarke, 
- [ ] @charlesHenryPATARD, 
- [ ] @ksdj, 
- [ ] @xinyuwu, 
- [ ] @jpenning, 
- [ ] @willrogers, 
- [ ] @claudio-rosati, 
- [ ] @gciotto, 
- [ ] @wmoore28

charlesHenryPATARD, 2017-08-29:
Ok for us.

willrogers, 2017-08-29:
Several things for me:

* Right now we have a build problem.  I will try to resolve this promptly since it appears it is not affecting other sites
* I have a second proposal for #2205, on which I have stopped working because we haven't been able to build
* I would like to be able to include databrowser 3 in my 4.5 build.  Is this possible?
* Ideally I would like to be able to include Display Builder in my 4.5 build as well as BOY.  Is this possible?  I'm not sure how to do this at present, but a p2 repo for Display Builder might be the easiest way to do this

Cheers,
Will

berryma4, 2017-08-29:
@willrodgers, I believe ESS has their product with data browser 3 and display builder. @claudio-rosati

kasemir, 2017-08-29:
Yes. Similarly, the SNS product includes
 * BOY
 * Display Builder
 * Data Browser 3
 * databrowser3.opiwidget (for data browser in BOY)
 * databrowser3.bobwidget (for data browser in display builder)

You could use just
 * BOY
 * Data Browser 3
 * databrowser3.opiwidget (for data browser in BOY)
if you don't want to have the display builder.
On beamlines, we now use the above full set but leave the 'top displays' menu of the display builder empty for those beamlines that are still primarily using BOY.
Or the other way around: We leave the 'top displays' menu of BOY empty on beamlines that are already updated to the display builder.

willrogers, 2017-08-29:
I would like to choose one of those two configurations.  It may be a little confusing to have both BOY and Display Builder, but I would like to start assessing Display Builder as soon as possible.

However, it is not clear to me how to build the Display Builder code into the DLS cs-studio product.

kasemir, 2017-08-29:
https://github.com/ControlSystemStudio/org.csstudio.sns/tree/master/build has examples for cloning the display builder repo with the rest of CSS and compiling it into the SNS product.

jpenning, 2017-08-31:
Hi Eric,

Ok for DESY J

Thanks for asking,

Jörg.

<https://github.com/ControlSystemStudio/cs-studio/issues/2267> , or mute the 
thread

willrogers, 2017-08-31:
Pull request open here: https://github.com/ControlSystemStudio/cs-studio/pull/2272

nickbattam, 2017-09-04:
Pull request open here: [maven-osgi-bundles #49](https://github.com/ControlSystemStudio/maven-osgi-bundles/pull/49)

closed 2019-04-25 by berryma4