ControlSystemStudio/cs-studio #2268: Various 4K hidpi issues

boryto, 2017-08-29:
On 4k screens,  issues with scaling of some components arise, specially when using bigger fonts. 
The first to be solved is the log entry dialogue:
[image]
Hide/Show details button almost hidden behind all other elements. 

This issue addresses all hidpi problems.

boryto, 2017-08-30:
With Neon release, Eclipse has native support for scaling images on hidpi monitors.   https://www.eclipse.org/eclipse/news/4.6/platform.php#swt-autoscale 
Scaling is realized with SWT scaling  - the basic principle is re-sizing existing icon images and taking  images with @2x or @1.5x (200% or 150% scaling) at zoom levels to get crisp images. 

While this looks good on most of the Cs-studio components, it represents problem on BOY opi's. Specially existing ITER opi's, which are already designed for 4K. The core of the problem is that ITER has already developed OPI's which are designed for 4K. This opi's already contain 4k images, and opi's are designed with borders over 3000px (width) . If we use native scaling,  images and borders are also scaled. This causes the existing opi's to look bad (over-scaled) when scaling to 150% or 200%.

One way to fix this issue would be to only disable this kind of scaling for boy, but this support is build direct into Image class, and there is no parameter on the Image or ImageDescriptor class to disable this scaling. There are also some icons (alarm table widget), that are used in boy and in alarm table. So it doesn't look so good. 

Seems that only reasonable way to deal with the problem is to create bigger icons, and make the preference to enable them.  We could use the same suffix for images @2x. If preference is true, @2x icon images would be used. If not, normal images would take place. So we have full control over icon images. This would require a helper class (as ImageUtil)  to be used at many places in code.

Any other ideas?

kasemir, 2017-08-30:
Can you attach screenshots of some section in your displays with older CSS vs. based on neon, on normal and 4K display?

boryto, 2017-09-05:
Attached screenshots are displays with no scaling, and with 150% scaling. When no scaling is used the Opi's look the same as on Mars. So the diffrence is only between not scaled and scaled. 

The editor icons on scaled screen look OK, but the OPI is to big, as the images and borders are bigger.

[image]
[image]
[image]
[image]
[image]
[image]

kasemir, 2017-09-05:
Thanks for the screenshots. Let me rephrase to assert that I got it:
With SWT autoscaling as included in Neon, everything is scaled for high DPI monitors. Those who never did anything to prepare for high DPI will be fine. You can fine-tune as described in https://www.eclipse.org/eclipse/news/4.6/platform.php#swt-autoscale-tweaks .

ITER, however, has used high DPI monitors for some time already, your images are already 1.5 to 2x enlarged, and those must _not_ be auto-scaled once more.
If I understand https://www.eclipse.org/eclipse/news/4.6/platform_isv.php#high-dpi-icons correctly, you could rename all your existing images by adding `@2x` to the name, and then these images would not be auto-scaled once more.

Is that the solution? For your existing displays, you rename images to `..@2x`? For icons, plain icons will be auto-scaled, and if we want to add higher res icons, we have both the plain and the `..@2x` version?
The remaining work: While `org.eclipse.jface.resource.ImageDescriptor#createFromURL(URL)` already handles the `..@2x` convention, our code in BOY or elsewhere that opens an image by other means and just passes an `InputStream` on to the `Image`, we need to handle the `..@2x` ourselves?

boryto, 2017-09-08:
Hi, yes the native way from fullHD to 4k would be using SWT scaling, where everything is scaled for factor 2x. So users that didn't change the OPI's are in theory OK. 
But you have be aware that for ex. position h:600, and w: 800 is displayed as (1200, 1600).  This goes for everything, not just images (labels, buttons, rectangles....).  So just changing the names of the images is not an option. 
I also tried to change the name of the picture, but it scales anyway. I didn't investigate this option further. 

Scaling is handled in the `org.eclipse.swt.graphics.Image`. This class is behind every image that CSS uses, so no need to do any other work in CSS in case of using SWT scaling. Everything is scaled except fonts. 

As images are resized, they look bad (pixel edges) if they don't have @2x suffix. So if we want to have nice hidpi icons in CSS, we must create bigger icons with @2x anyway. 

ITER doesn't want to scale OPI, but just want's bigger icons. So we are thinking about solution, where we disable autoscaling and just force CSS to use bigger icons (in case they exist). As eclipse already provides @2x icons in jars, we just have to create the CSS @2x versions. 

Forcing the bigger icons would be enabled in .ini file.

For forcing icons, there is an option to create own `org.eclipse.swt.graphics.Image` class and use it in our product. As this is kind of hack, it means only changing the code at one place, and there is no affect on other code, and can be used only by ITER. 

The other option is to insert usage of this preference in places where plugins load images. As I researched if so far, there is no single CSS class that is used for image loading, so we would have to change code at many places.  For eclipse part we can change this by declaring new workbench images, so this can be handled quite nicely.

kasemir, 2017-09-08:
Related: JavaFX also seems to honour the same type of naming.
If you load `icon-name.png` it will check if there's an alternate `icon-name@2x.png` or `icon-name@3x.png` in the same location.

http://dlsc.com/2017/08/29/javafx-tip-27-hires-retina-icons/

So CSS `@2x` icons created now for SWT would still be useful with JFX.

boryto, 2017-09-11:
Bad layout on Service button dialog:
[image]

fix:
[image]