ControlSystemStudio/cs-studio #2270: TableViewer row not rendered properly (collapsed)

boryto, 2017-08-29:
Rows are not properly rendered on No RHEL7 with GNOME-3 
The issue is present in alarm table widget, and on PV table and auto-complete.

While forcing the cs-studio to use the gnome-2 (export SWT_GTK3=0) solves the issue for the PV table and autocomplete, the issue on the alarm table widget remains. 

Alarm table widget:
[image]

 PV table:
[image]

boryto, 2017-08-29:
Looks like there is a SWT problem with GridLayout using grabExcessVerticalSpace. Workaround for this is to set the parameter to false and pass height hint to layout.  This fixes the alarm table widget.

kasemir, 2017-08-29:
We do need to force GTK2 (export SWT_GTK3=0) because Java 8 on Linux only supports GTK2 for Java FX, and there are now some JavaFX-based components: Logging configuration, standalone displays view, display builder, databrowser3, ..

With Java 9, it'll be the other way round: Must use GTK 3.

boryto, 2017-08-30:
Closing the issue for now, as GTK-2 solves the issue on PV table, and we need to use GKT-2 on Linux. If problem remains on Java9, we need to reopen it.

closed 2017-08-30 by boryto