ControlSystemStudio/cs-studio #2273: Menu Button should not accept all VTypes for label

css-iter, 2017-08-30:
The MenuButton has change listener set to change the label of the button if the value of the associated PV changes. I guess this may be useful in some cases for setting the labels from the control system.

However, the widget is not particularly selective about the VType it accepts as an input, which can change the label to something like "VTable[2x11, [Key, Value]]".

I think that if we allow the value of the label to be changed at all base on the associated datasource value, it should be limited only to Scalar values, as no other values make any sense.

I will submit a PR to that effect.

kasemir, 2017-08-30:
Well, going forward everything should try to handle not just Channel Access but also pvAccess data. Unclear what it should show for a VTable or VImage, since attaching those PVs to a button doesn't make sense in the first place. But then showing "VTable.." so you know that you picked the wrong PV seems better than not showing anything.

css-iter, 2017-09-01:
The problem that we at ITER have is that the button is connected to a Beast data source (so that it's color/status reacts to an alarm state). Which is always a VTable. In this case this is a problem.

css-iter, 2017-09-07:
Merged into the main product.

closed 2017-09-07 by css-iter