ControlSystemStudio/cs-studio #2276: BOY Combo should ignore mouse wheel

kasemir, 2017-08-30:
Our users tend to use the mouse wheel to scroll displays up/down.

If the mouse happens to be on top of a combo box, the scroll wheel events are captured by the widget and written to the PV.

This is unexpected. Writing to a PV, i.e. causing something to happen usually requires pressing "Enter" or deliberately clicking on something.

--> Combo should ignore the mouse wheel.

willrogers, 2017-08-30:
I agree.

kasemir, 2017-08-30:
Thank you, finally somebody agrees with me :-)

The relevant code is in the `comboSelectionListener` of `org.csstudio.opibuilder.widgets.editparts.ComboEditPart`.

It can be invoked in several ways:

 * Clicks on an item in the combo bo -->  `e.stateMask == SWT.BUTTON1`
 * Roll mouse wheel, happen to be on top of combo -->  `e.stateMask == 0`
 * Activate the widget, press SPACE to open the drop down list of items, cursor keys to select item, then press SPACE or ENTER  -->  `e.stateMask == 0`

I can register a designated `combo.addMouseWheelListener(..)'. It will be invoked _in addition_ to the selection listener, it won't capture the mouse wheel events.

Key presses like CTRL are not consistently reflected in the `stateMask`.

==>  Cannot ignore just the mouse wheel. Cannot distinguish mouse wheel from keyboard as source of selection. Could make it work with _only_ mouse clicks, ignoring both mouse wheel (good) and keyboard (bad for people who have no mouse or insist in being 10x faster by only using the keyboard).

willrogers, 2017-08-30:
We disabled navigation of OPIs using the keyboard because:

* it led to unexpected behaviour, as above
* it didn't work properly anyway

See #2038.  So I don't think the keyboard should do anything anyway.  If you want to support keyboard navigation properly in Display Builder, I think you need to do it carefully (perhaps ensuring no widget is selected by default is good enough).

PR #2277 (2017-09-05): Combo: Ignore mouse wheel

Fixes #2276
closed 2017-09-05 by berryma4