ControlSystemStudio/cs-studio #2284: Can't delete Databrowser trace

willrogers, 2017-09-14:
Steps to reproduce:

* Delete PV name in text box and press enter
* You get a pop-up saying 'Cannot change item name...'
* Press OK

You can no longer delete that trace using the context menu.

willrogers, 2017-09-14:
Generated exception:

```
2017-09-14 15:26:07.983 WARNING [Thread 1] org.csstudio.swt.rtplot.undo.UndoableActionManager (execute) - Action failed: Delete Item
java.lang.RuntimeException: Not running 
	at org.csstudio.trends.databrowser2.model.PVItem.stop(PVItem.java:327)
	at org.csstudio.trends.databrowser2.model.Model.removeItem(Model.java:474)
	at org.csstudio.trends.databrowser2.propsheet.DeleteItemsCommand.run(DeleteItemsCommand.java:51)
	at org.csstudio.swt.rtplot.undo.UndoableActionManager.execute(UndoableActionManager.java:57)
	at org.csstudio.trends.databrowser2.propsheet.DeleteItemsCommand.<init>(DeleteItemsCommand.java:43)
	at org.csstudio.trends.databrowser2.propsheet.DeleteItemsAction.run(DeleteItemsAction.java:85)
	at org.eclipse.jface.action.Action.runWithEvent(Action.java:473)
	at org.eclipse.jface.action.ActionContributionItem.handleWidgetSelection(ActionContributionItem.java:565)
	at org.eclipse.jface.action.ActionContributionItem.lambda$4(ActionContributionItem.java:397)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:84)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:5227)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1340)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:4561)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:4151)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$4.run(PartRenderingEngine.java:1121)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1022)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:150)
	at org.eclipse.ui.internal.Workbench$5.run(Workbench.java:693)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:610)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:148)
	at org.csstudio.utility.product.Workbench.runWorkbench(Workbench.java:103)
	at org.csstudio.startup.application.Application.startApplication(Application.java:265)
	at org.csstudio.startup.application.Application.start(Application.java:119)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:196)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:388)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:243)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:673)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:610)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1519)
	at org.eclipse.equinox.launcher.Main.main(Main.java:1492)
```

willrogers, 2017-09-14:
In fact, is there any reason to change the item name at all?

kasemir, 2017-09-14:
The item name is the PV name. Sometimes you do want to change that as a shortcut to adding a new item, then removing the old item.

It _is_ possible to 
1) Add pv sim://sine
2) Then rename that to sim://ramp

The bug you see only happens when we try to rename a PV to "", the empty name

-> I suggest to catch that, and just leave the original name in place.

kasemir, 2017-09-14:
Fixed it in Databrowser3

willrogers, 2017-09-14:
Can you link to the commits so I can see the changes?

kasemir, 2017-09-14:
It should show above my comment: https://github.com/kasemir/org.csstudio.display.builder/commit/46b1f82094536cfd680494b6e00b22195e5f36b1

kasemir, 2017-09-14:
Ah, the link shows on the github web page because the commit referenced this issue.
If you only have the email from github, you wouldn't see the link in there.

willrogers, 2017-09-14:
Oh, sorry, got it.

closed 2017-09-29 by willrogers