ControlSystemStudio/cs-studio #2290: How to troubleshoot PVStructureHelper.getVType()?

mdavidsaver, 2017-09-20:
In working on a PVAccess server I'm faced with a situation where (I think) something is going wrong with the unpacking of PVStructure into a VType.  I suspect that this is something unfortunate like eg. my providing PVLong when PVInt is needed..

All I see is a generic error about "null value".  Can hints on how I can get more information about where the mis-match is?

In this case I"m trying to provide an NTNDArray (cf. class VImageForNTNDArray).

kasemir, 2017-09-20:
Are you creating a display builder screen, where an Image widget is getting the error?

The VImageForNTNDArray code has been tested with the image created by

1) The 'ntndarrayServer' in https://github.com/kasemir/EPICSV4Sandbox, which is based on David Hickin's code, but changed the data type to try different data types.

2) The python code in https://github.com/kasemir/org.csstudio.display.builder/tree/master/org.csstudio.display.builder.model/examples/image_server that serves some *.png file as an NTNDArray suitable for the display builder image widget

kasemir, 2017-09-20:
Usually, the log file (workspace directory/.metadata/.log) would have the stack trace for null pointer exceptions.
The "null value" likely just means that something wants to display null on the screen, doesn't tell us where/why we ended up with a null.

mdavidsaver, 2017-09-20:
> Are you creating a display builder screen, where an Image widget is getting the error?

This is still BOY Intensity Graph w/ cs-studio nsls2 4.3.1.  This is mostly inertia on my part.  And the fact that I'm manually replacing CAJ w/ a patched version to get name lookup over TCP.  So upgrading is more tedious than usual.

> The VImageForNTNDArray code has been tested with the image created by

I'm aware of this.  You may assume from this question that I'm doing something new and different which I may be boring you with in two weeks.

> Usually, the log file (workspace directory/.metadata/.log

No traces which look likely.

kasemir, 2017-09-20:
The BOY intensity graph doesn't support VImage.
It can only handle waveforms, i.e. VNumberArray. pva://my_image/value might work, which fetches that element of the NTNDArray structure.

mdavidsaver, 2017-09-20:
Ha, well that's inertia for you...  And hey, if I scroll down there's a widget named "VImage".

closed 2017-09-20 by mdavidsaver