ControlSystemStudio/cs-studio #2311: CS-Studio & Java 9

claudio-rosati, 2017-10-23:
Hello all,

I've just tried to install Java 9 on my test VM (CentOS 7.4) and run CS-Studio (built using Java 8) to text if it runs with or without problems, and it happens to have problems (see later on). Have you tried to do the same on other platforms/OSes?

```
[claudiorosati@localhost Downloads]$ java -version
java version "9.0.1"
Java(TM) SE Runtime Environment (build 9.0.1+11)
Java HotSpot(TM) 64-Bit Server VM (build 9.0.1+11, mixed mode)
[claudiorosati@localhost Downloads]$ css
Oct 23, 2017 1:39:43 PM org.csstudio.diirt.util.Activator start
INFO: Starting diirt.util
Oct 23, 2017 1:39:43 PM org.csstudio.diirt.util.RegisterDatasource registerDatasource
INFO: register Datasource:beast
Oct 23, 2017 1:39:43 PM org.diirt.util.config.Configuration getFile
INFO: Loading /home/claudiorosati/.diirt/datasources/datasources.xml
Found bundle org.diirt.datasource_3.1.7.20171019210847 [646]
Found bundle org.eclipse.osgi.internal.framework.BundleContextImpl@54a5e10b
Found bundles [Lorg.osgi.framework.Bundle;@65cf0b66
Found bundleentry://494.fwk181260145/META-INF/services/org.diirt.datasource.DataSourceProvider in org.csstudio.archive.diirt.datasource_1.0.1.201710191917 [494]
Providers [org.csstudio.archive.diirt.datasource.ArchiveDataSourceProvider]
Provider instance org.csstudio.archive.diirt.datasource.ArchiveDataSourceProvider@1705ad2
Found bundleentry://648.fwk181260145/META-INF/services/org.diirt.datasource.DataSourceProvider in org.diirt.datasource-file_3.1.7.20171019210843 [648]
Providers [org.diirt.datasource.file.FileDataSourceProvider]
Provider instance org.diirt.datasource.file.FileDataSourceProvider@17f28134
Found bundleentry://652.fwk181260145/META-INF/services/org.diirt.datasource.DataSourceProvider in org.diirt.datasource-loc_3.1.7.20171019210843 [652]
Providers [org.diirt.datasource.loc.LocalDataSourceProvider]
Provider instance org.diirt.datasource.loc.LocalDataSourceProvider@635a7663
Found bundleentry://654.fwk181260145/META-INF/services/org.diirt.datasource.DataSourceProvider in org.diirt.datasource-sim_3.1.7.20171019210842 [654]
Providers [org.diirt.datasource.sim.SimulationDataSourceProvider]
Provider instance org.diirt.datasource.sim.SimulationDataSourceProvider@1888a927
Found bundleentry://655.fwk181260145/META-INF/services/org.diirt.datasource.DataSourceProvider in org.diirt.datasource-sys_3.1.7.20171019210842 [655]
Providers [org.diirt.datasource.sys.SystemDataSourceProvider]
Provider instance org.diirt.datasource.sys.SystemDataSourceProvider@e6bb57a
Found bundleentry://663.fwk181260145/META-INF/services/org.diirt.datasource.DataSourceProvider in org.diirt.support.diirt-ca_3.1.7.20171019210846 [663]
Providers [org.diirt.support.ca.JCADataSourceProvider]
Provider instance org.diirt.support.ca.JCADataSourceProvider@3fd96c80
Found bundleentry://665.fwk181260145/META-INF/services/org.diirt.datasource.DataSourceProvider in org.diirt.support.diirt-pva_3.1.7.20171019210846 [665]
Providers [org.diirt.support.pva.PVADataSourceProvider]
Provider instance org.diirt.support.pva.PVADataSourceProvider@6d236021
Oct 23, 2017 1:39:44 PM org.csstudio.diirt.util.RegisterDatasource registerDatasource
INFO: register Datasource:archive
2017-10-23 13:39:45.940 SEVERE [Thread 1] org.csstudio.startup.application.Application (start) - Application Main Loop Error
org.eclipse.e4.core.di.InjectionException: java.lang.NoClassDefFoundError: javax/annotation/PostConstruct
	at org.eclipse.e4.core.internal.di.InjectorImpl.internalMake(InjectorImpl.java:386)
	at org.eclipse.e4.core.internal.di.InjectorImpl.make(InjectorImpl.java:294)
	at org.eclipse.e4.core.contexts.ContextInjectionFactory.make(ContextInjectionFactory.java:162)
	at org.eclipse.e4.ui.internal.workbench.swt.E4Application.createDefaultHeadlessContext(E4Application.java:490)
	at org.eclipse.e4.ui.internal.workbench.swt.E4Application.createDefaultContext(E4Application.java:504)
	at org.eclipse.e4.ui.internal.workbench.swt.E4Application.createE4Workbench(E4Application.java:203)
	at org.eclipse.ui.internal.Workbench$5.run(Workbench.java:632)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:610)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:148)
	at org.csstudio.utility.product.Workbench.runWorkbench(Workbench.java:103)
	at org.csstudio.startup.application.Application.startApplication(Application.java:265)
	at org.csstudio.startup.application.Application.start(Application.java:119)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:196)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:388)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:243)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:564)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:673)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:610)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1519)
	at org.eclipse.equinox.launcher.Main.main(Main.java:1492)
[... 31 lines cut ...]
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:388)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:243)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:564)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:673)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:610)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1519)
	at org.eclipse.equinox.launcher.Main.main(Main.java:1492)
Caused by: java.lang.NoClassDefFoundError: javax/annotation/PostConstruct
	at org.eclipse.e4.core.internal.di.InjectorImpl.inject(InjectorImpl.java:151)
	at org.eclipse.e4.core.internal.di.InjectorImpl.internalMake(InjectorImpl.java:375)
	... 25 more
Caused by: java.lang.ClassNotFoundException: javax.annotation.PostConstruct cannot be found by org.eclipse.e4.core.di_1.6.1.v20160712-0927
	at org.eclipse.osgi.internal.loader.BundleLoader.findClassInternal(BundleLoader.java:410)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:372)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:364)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.loadClass(ModuleClassLoader.java:161)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:496)
	... 27 more

2017-10-23 13:39:46.119 INFO [Thread 26] org.csstudio.diirt.util.RegisterDatasource (deregisterDatasource) - deregister Datasource:archive
2017-10-23 13:39:46.120 INFO [Thread 26] org.csstudio.diirt.util.RegisterDatasource (deregisterDatasource) - deregister Datasource:beast
```

nickbattam, 2017-10-23:
Does it run if you compile it with Java9?

kasemir, 2017-10-23:
Good news is that Eclipse 4.7.1a does run with Java 9. We might need to update our target platform (using that term loosely, since we don't have any) to Eclipse 4.7.1.a and remove remaining access to what's now internal packages in the CS-Studio code.

Meanwhile, is there any improvement when trying to run with `-vmargs  --add-modules=ALL-SYSTEM`?

willrogers, 2017-11-01:
We've agreed to look at this at DLS.

We may not do much work on this right now, since it seems that Eclipse itself may not be quite ready.

willrogers, 2017-11-01:
From Kay:

> It is a bigger change to Java itself, so need to move from Java 8 & Eclipse 4.6 to Java 9 & Eclipse 4.7.1a.
Can't update just Java, since Eclipse 4.6 won't work with that.
Could update just to Eclipse 4.7.(before 1a) while staying with Java 8 as a middle step, not sure if that makes it easier or just wastes time.

> Java 8 will continue to be be supported for at least a year.
On the other hand, we've just started the VType update to use 'default' in interfaces.
Java 9 offers API for JavaFX features that used to be private.
So waiting a full year would limit options for sharing code with new developments.

> Maybe start to look at what's to do, but wait for Eclipse 4.7.2
( https://wiki.eclipse.org/Oxygen/Simultaneous_Release_Plan#Oxygen.2 )
because that would be the first "real" release of Eclipse for Java 9?

willrogers, 2018-01-19:
Tycho is still discussing java 9: https://bugs.eclipse.org/bugs/show_bug.cgi?id=514471

kasemir, 2018-01-19:
Thanks for monitoring the progress!

It's a little underwhelming.  While 4.7.1a includes the J9 support, https://www.eclipse.org/eclipse/news/4.7.1a/, I couldn't get the 'modules' to fully work.
The basic 'requires' and 'exports' is understood, you can access classes in the IDE based on the module file. But the 'requires transitive' which you really want for a module hierarchy seemed to be ignored; had to always import every individual module.

Of course we just use Eclipse as a free tool, so we can't complain about missing features.

berryma4, 2018-04-06:
We should be able to move to GTK3 in java 9
http://openjdk.java.net/jeps/283

shroffk, 2020-07-23:
Move to java 11 is complete however gtk3 support still has issues.
I will close this ticket since it is primarily aimed and jdk9 and we can create a dedicated issue ticket about gtk3 support

closed 2020-07-23 by shroffk