ControlSystemStudio/cs-studio #2314: Archive configuration import - next sequence ID

utzeln, 2017-10-25:
During the import the first ID assigned is usually 'select max(id) from... ' + 1, and this for smplt_eng, chan_grp, channel tables.

Another option will be to let the RDB assign the sequence id automatically.

In the case of Postgres it is possible to define for a sequence generator minimum, maximum and starting values:

create sequence smpl_eng_engid start 5000 minvalue 5000 maxvalue 15000;

In this case, the first ID assigned by the sequence generator will be 5000 and not 1 by default.

This allows to manage unique IDs on distributed databases in order to provide a unified view on the channel samples - making sure that each channel has a unique ID over multiple databases, and not having in all databases a channel 1, channel 2... And this without using the UUID data type which is not cast properly by the tools.

The current workaroung is to create dummy raw with id = 5000 and let the 'select max(id) from... ' + 1 works.

kasemir, 2017-10-25:
You're right.
In org.csstudio.archive.config.rdb.SQL, smpl_eng_next_id is for now always "SELECT MAX(..)".
It could be changed into
```
if (dialect == Dialect.PostgreSQL)
    smpl_eng_next_id = "..get next ID from sequence..";
else
    smpl_eng_next_id = "SELECT MAX(eng_id) FROM " + schema + "smpl_eng";
```

I don't get the "distributed databases" argument. If it's properly distributed, i.e. a cluster of several servers that appear as one, doesn't that still support transactions and referential integrity? So another way to assert unique IDs would be to verify correct use of transactions:
 1) Start transaction
 2) next_id = SELECT MAX(id) + 1
 3) INSERT .. with next_id
 4) Close transaction

That would have the advantage of working with any database dialect, whereas the dialect-specific `smpl_eng_next_id` might in one case be a plain statement, in another case a stored procedure call, so you end up with more "if (dialect == .." checks.

utzeln, 2017-10-25:
Context: We need to setup temporary control and server rooms with "local" databases in each building. In the contruction phase, we can not rely on network and we need to support network shutdown of maximum 1 week. That is why we cannot setup a cluster over multiple locations. 
So one solution is to have local databases in "standalone mode" and to setup a central virtual database with proxy tables on the remote locations.  This using Postgresql Foreign Data Wrapper which allows the creation of foreign tables in a PostgreSQL database that are “proxies” for some other data source
When a query against a foreign table is done, the Foreign Data Wrapper will query the external data source and return the results as if they were coming from a table in the local database.
This way we can have users in remote location working in standalone even if the network is down, but also users in the central location being able to plot any PVs from the remote locations.
But during the import of the archive configuration in the remote location, the IDs generates shall be unique from the central point of view.

utzeln, 2017-10-25:
So perhaps for the IDs we could have 2 modes: 1 managed by the RDB and in this case the INSERT INTO setup all the values except the ID, and the other one which manage the IDs in the Java code via the MAX(id) function.

kasemir, 2017-10-25:
Your explanation makes it less clear.
If the databases are disconnected, isn't the only solution your original workaround?
Pre-populate database 1 with a dummy engine ID of "5000", database 2 with a dummy engine ID of "6000", database 3 with a dummy engine ID of "7000" and so on, so they each create new IDs from that starting point on without overlap (may need to allow more than 1000 per database, but you get the idea).

utzeln, 2017-10-25:
With on database 1:
`create sequence smpl_eng_engid start 5000 minvalue 5000 maxvalue 15000;`
and on database 2:
`create sequence smpl_eng_engid start 25000 minvalue 25000 maxvalue 35000;`
and letting the sequence generator doing the job we do not need dummy raws like:
`INSERT INTO smpl_eng(name, descr, url) values('xxx', '', '');`
to force the first max(id) on each database 1 and 2.

For the channels, creating dummy `yyy` channels will be visible in the plot search view.

But we could also remove the dummy raw after the first import.

kasemir, 2017-10-25:
Again, I'm not opposed to you implementing support for Postgres/Oracle sequences or MySQL 'auto' IDs. This likely requires `if (dialect ==` in several places, because both statement and how to execute it will be different.

On the other hand, it looks like you can already get all you need by simply inserting dummy values with a high enough ID into each database. Only actual downside would be dummy channel names visible in the channel search. I'd pick a name like "___internal1" to make it obvious that it's not a real channel name, and continue with my day job.

utzeln, 2017-10-26:
OK - we keep the workaround as the temporary control room setup is just that - temporary. And thanks for the naming convention of the dummy channel name.

closed 2017-10-26 by utzeln