ControlSystemStudio/cs-studio #2317: Databrowser: save that I don't want to save changes

willrogers, 2017-10-31:
This might be me not understanding correctly.

Use case: I have saved `.plt` files that I don't want to save changes to, because they are configured as I want them. Sometimes I will zoom and pan etc, but I don't then want to be prompted to save them when I close CS-Studio.

There is a 'Save Changes' option in the misc properties tab. If I untick this, then I won't be prompted to save the file. But that means that I can't save this option into the file. When I open the file again, it will be saveable.

So what I want is for the 'Save changes' option to be saved into the file. Is this possible? Or have I misunderstood somehow?

kasemir, 2017-10-31:
You need to use "Save As" to save the ignore-changes option.
You can save into the existing file, just say "Yes" in the "Do you want to replace existing file" dialog.

kasemir, 2017-10-31:
From the online help:
```
Save Changes: ... If you still want to save changes, use File/Save As to save the settings into a new or even the original *.plt file.
```

willrogers, 2017-10-31:
Perfect, thanks.

closed 2017-10-31 by willrogers