ControlSystemStudio/cs-studio #2322: Proper way of handling custom message.properties.

boryto, 2017-11-15:
Hi, 
we have the following scenario: We want to change the StartEnd_EndTime_TT message
defined in org.csstudio.apputil.ui/src/org/csstudio/apputil/ui/time/messages.properties.

StartEnd_EndTime_TT=Enter end time (-2month -5days YYYY/MM/DD HH:MM:SS)
As we use ISO timestamp formatting, this would change to (-2month -5days YYYY-MM-DDTHH:MM:SSZ).

Is there any mechanism like preferenceCustomization extension where you can override messages defined in plugins. The new values of the messages would be defined in your own product, as a global messages.properties.  This is actually not a translation, it is override. 
If not, what is the preferred way for handling this kind of situations. We don't want to change the default value. 

TNX

kasemir, 2017-11-15:
For what it's worth, I think it's wrong to customize the formatting. See also https://github.com/shroffk/phoebus/issues/86

But too late, you want your own time format.
How about updating the text to be generic, just
```
StartEnd_EndTime_TT=Enter end time
```

Or maybe you can create your own locale, pass that via the "-nl ITER" command line parameter and create a `messages_ITER.properties` file, which I believe can also be in a site-specific fragment.

willrogers, 2017-11-15:
But then the format @boryto wants looks to be what I would consider the standard format to be.

I'm not sure that this change would bother many people, as it goes from nearly-standard to even-more-standard.

Where would I see this change if it happened?

boryto, 2017-11-16:
This message refers to the tool-tip on Start/End Time dialog, tool-tip on the databrowser time input box  and autocomplete message on log table query. 

Just changing this into ISO time would be wrong as it will propose wrong time format. 
As it looks there is no standard way, the easiest option wold be to leave the absolute time out of the suggestion, but this will probably not be good for all users. 

We will check the option with ITER's language. 
One option (maybe the best) is also  to inject/replace the chosen formatting from the Timestamp formats into the message string.

Thank you for your suggestions.

closed 2018-01-23 by boryto