ControlSystemStudio/cs-studio #2339: Databrowser: precision of annotation values

willrogers, 2017-12-05:
Operator request:

When you add an annotation to the databrowser, you have no control of the precision of the value that is rendered. This means that when you're trying to show differences sometimes the annotation doesn't show that there is a difference at all.

Could we provide some way to specify precision?

kasemir, 2017-12-05:
The precision is based on the current axis labels.

[image]

Zoom "in", and you get a higher precision:

[image]

It is, however, just like the axis setting.
Maybe it should use the 'crosshair' precision, which is basically "one more digit":

[image]

willrogers, 2021-04-13:
Fixed with #2674

closed 2021-04-13 by willrogers