ControlSystemStudio/cs-studio #2340: Exception thrown when adding annotation to databrowser

willrogers, 2017-12-05:
Frequently, when I add an annotation to the databrowser, I get the following exception:

```
2017-12-05 11:17:08.254 SEVERE [Thread 44] org.csstudio.logging.PluginLogListener (logging) - Invalid thread access
org.eclipse.swt.SWTException: Invalid thread access
        at org.eclipse.swt.SWT.error(SWT.java:4533)
        at org.eclipse.swt.SWT.error(SWT.java:4448)
        at org.eclipse.swt.SWT.error(SWT.java:4419)
        at org.eclipse.swt.widgets.Widget.error(Widget.java:483)
        at org.eclipse.swt.widgets.Widget.checkWidget(Widget.java:421)
        at org.eclipse.swt.widgets.Widget.getData(Widget.java:509)
        at org.eclipse.ui.internal.ide.handlers.PagePropertyTester.test(PagePropertyTester.java:36)
        at org.eclipse.core.internal.expressions.Property.test(Property.java:58)
        at org.eclipse.core.internal.expressions.TestExpression.evaluate(TestExpression.java:100)
        at org.eclipse.core.internal.expressions.CompositeExpression.evaluateAnd(CompositeExpression.java:53)
        at org.eclipse.core.internal.expressions.WithExpression.evaluate(WithExpression.java:75)
        at org.eclipse.core.internal.expressions.CompositeExpression.evaluateOr(CompositeExpression.java:68)
        at org.eclipse.core.internal.expressions.OrExpression.evaluate(OrExpression.java:22)
        at org.eclipse.core.internal.expressions.ReferenceExpression.evaluate(ReferenceExpression.java:71)
        at org.eclipse.ui.internal.services.EvaluationReference.evaluate(EvaluationReference.java:72)
        at org.eclipse.ui.internal.services.EvaluationReference.evaluate(EvaluationReference.java:97)
        at org.eclipse.ui.internal.services.EvaluationService.requestEvaluation(EvaluationService.java:294)
        at org.eclipse.ui.internal.handlers.DirtyStateTracker.update(DirtyStateTracker.java:45)
        at org.eclipse.ui.internal.handlers.DirtyStateTracker.propertyChanged(DirtyStateTracker.java:107)
        at org.eclipse.ui.part.WorkbenchPart.firePropertyChange(WorkbenchPart.java:120)
        at org.csstudio.trends.databrowser2.editor.DataBrowserEditor.setDirty(DataBrowserEditor.java:460)
        at org.csstudio.trends.databrowser2.editor.DataBrowserEditor$2.changedAnnotations(DataBrowserEditor.java:269)
        at org.csstudio.trends.databrowser2.model.Model.setAnnotations(Model.java:846)
        at org.csstudio.trends.databrowser2.ui.Controller$3.changedAnnotations(Controller.java:379)
        at org.csstudio.trends.databrowser2.ui.ModelBasedPlot$2.lambda$2(ModelBasedPlot.java:127)
        at java.util.Optional.ifPresent(Optional.java:159)
        at org.csstudio.trends.databrowser2.ui.ModelBasedPlot$2.changedAnnotations(ModelBasedPlot.java:127)
        at org.csstudio.swt.rtplot.internal.Plot.fireAnnotationsChanged(Plot.java:1254)
        at org.csstudio.swt.rtplot.internal.Plot.addAnnotation(Plot.java:508)
        at org.csstudio.swt.rtplot.undo.AddAnnotationAction.run(AddAnnotationAction.java:32)
        at org.csstudio.swt.rtplot.undo.UndoableActionManager.execute(UndoableActionManager.java:57)
        at org.csstudio.swt.rtplot.internal.PlotProcessor.lambda$2(PlotProcessor.java:333)
        at java.util.concurrent.ForkJoinTask$RunnableExecuteAction.exec(ForkJoinTask.java:1402)
        at java.util.concurrent.ForkJoinTask.doExec(ForkJoinTask.java:289)
        at java.util.concurrent.ForkJoinPool$WorkQueue.runTask(ForkJoinPool.java:1056)
        at java.util.concurrent.ForkJoinPool.runWorker(ForkJoinPool.java:1692)
        at java.util.concurrent.ForkJoinWorkerThread.run(ForkJoinWorkerThread.java:157)
```

kasemir, 2017-12-05:
Yes, see that as well.

kasemir, 2017-12-05:
See commit for databrowser3, https://github.com/kasemir/org.csstudio.display.builder/commit/84c54741698961d018b3ddd78f99323645348fc0, that moves the operations to the UI thread.
Databrowser2 needs equivalent with `Display.....asynExec()`

willrogers, 2021-04-13:
I no longer see this.

closed 2021-04-13 by willrogers