ControlSystemStudio/cs-studio #2341: Update to caj 2.4.0

berryma4, 2017-12-05:
I'm interested in this bug fix:
https://github.com/epics-base/jca/commit/ecc761b5313579d8102169080b63beaaefefd756
I saw where after cs-studio was shutdown normally, if I did a netstat it was still hooked to a few IOCs.
IOCs complain that a stale client id is being used, which usually arises after rebooting IOCs.
Some older IOCs, this situation crashes the IOC.

 - update caj 2.4.0 in cs-studio maven-osgi-bundles
 - update caj 2.4.0 in diirt
 - update diirt in cs-studio

shroffk, 2017-12-05:
The issue should read
Update jca to 2.4.0

We had merged the jca/caj projects and will now release them as a single module...this should not require any code change but we can reduce the dependencies of modules that require caj.

Currenlty the 2.4.0 has been pre released as part of the epics7 release and is available in the sonatype staging repo. I hope to have it completely released within a week (once I get the green light for all the epics 7 java modules)

berryma4, 2017-12-05:
So we should also update pvaccess and pvdata?
pvaccess 5.1.0
pvdata 6.1.0

shroffk, 2017-12-05:
yes, 
no reason not to

berryma4, 2017-12-22:
diirt has a pull request:
https://github.com/diirt/diirt/pull/91
Could we get a release with this?

berryma4, 2017-12-22:
I've deployed jca 2.4.0 from staging, before the 2.4.1 release.  It looks to solve the issue.

berryma4, 2017-12-22:
Side note, I also saw the error from the archive appliance.  If this works, we should make sure that gets updated soon.

berryma4, 2018-01-30:
@shroffk can diirt be released now?