ControlSystemStudio/cs-studio #2342: ADL2BOY converter application

kasemir, 2017-12-05:
Implement command-line callable application for the ADL (MEDM) file converter.

When CSS includes the ADL converter, this currently only offers a GUI interface:
Select files in navigator, use context menu to convert (which also opens the result in the BOY editor), then copy *.adl files out of workspace.

Idea is to add a headless application to CSS, which will basically allow calling CSS with command-line options as follows to run the converter without the UI:

```
 -application org.csstudio.opibuilder.adl2boy.application /path/to/file1.adl  /path/to/file2.adl 
```

kasemir, 2017-12-05:
Assume you have a directory `displays` with `*.adl` files:
```
css  -nosplash -application org.csstudio.opibuilder.adl2boy.application displays/*.adl
```

This uses the Linux shell to expand the arguments.
On Windows you might have to use

```
css  -nosplash -application org.csstudio.opibuilder.adl2boy.application displays/file1.adl displays/file2.adl
```

claudio-rosati, 2017-12-06:
@kasemir 
is there a documentation of all the applications (and their parameters) that can be launched in this way?

kasemir, 2017-12-06:
Depends.
Just run it with "-application XXX" and you get an error message "unknown application XXX, available applications: .. long list". You then then look for each one of them, and find for example https://help.eclipse.org/mars/index.jsp?topic=%2Forg.eclipse.platform.doc.isv%2Fguide%2Fp2_director.html for the P2 applications that allow installing features from the command line.

closed 2017-12-06 by kasemir