ControlSystemStudio/cs-studio #2350: combo control write value

lotrus, 2017-12-15:
Hello,
We can’t set a PV value using Combo in CSS 4.5, We have tried with the BOY Example ChoiceWidgets.opi and it does not seems to work either.
We have no issue with CSS 4.1 , did anything changed?
Best

kasemir, 2017-12-15:
Which part doesn't work?

For `/BOY Examples/widgets/3_6_ChoiceWidgets.opi` to run, you need a record called `css:sine`, which you can for example get like this:

```
record(calc,"$(user):count") {
        field(SCAN,".1 second")
        field(INPA, "$(user):count.VAL")
        field(CALC, "A+1")
}

record(calc,"$(user):sine") {
          field(SCAN,".1 second")
        field(INPA, "$(user):count.VAL")
        field(CALC, "100*sin(A/10)")
        field(HIHI, "80")
        field(HHSV, "MAJOR")
        field(HIGH, "60")
        field(HSV, "MINOR")
        field(LOW, "-60")
        field(LSV, "MINOR")
        field(LOLO, "-80")
        field(LLSV, "MAJOR")
        field(HOPR,"125")
          field(LOPR,"-125")
}
```

Or read `/BOY Examples/softIOC.txt`

[image]

lotrus, 2017-12-15:
We don't have trouble with the period ones.  The control that doesn't work is the combo control associated with the PV loc://myChoice.  We can change the value from blue to grey with the other controls but not with Combo Box_1 (label) or Combo. We are using the CSS SNS 4.5.

kasemir, 2017-12-15:
Can't reproduce. I'm using the latest sources from the git master, and with both vtype.pv and pvmanager both combo boxes as well as the text input in the upper right corner work fine.
The tooltip on all widgets shows that loc://myChoice is of type VString

lotrus, 2017-12-15:
We are realizing that it's not working only with sns-css-4.5.0-win32.win32.x86.zip available (nightly build of SNS), the 64 bits is ok.

kasemir, 2018-04-03:
Same issue as #2413

PR #2415 (2018-04-04): Issue 2413 boy combo windows
 #2276 tried to prevent combo box value changes via the mouse-wheel-guard.

Turns out that doesn't work on Windows, where the combo box event doesn't include mouse button information, so there's no straight forward way to distinguish an honest 'click' by the user to select a combo item from a mouse wheel induced flurry of value changes.

Fixes #2350 (and duplicate #2413)
closed 2018-04-04 by berryma4