ControlSystemStudio/cs-studio #2355: add git submodules to org.csstudio.product

berryma4, 2017-12-20:
** this change may break your build **

To allow us to rebuild a specific tagged version, I would like to add maven-osgi-bundles, cs-studio-thirdparty, and cs-studio as git submodules to the org.csstudio.product git repo.

This will also allow us to remove the last jenkins job on cloudbees, used for org.csstudio.product deploys.

org.csstudio.product deploy will now be done when the repository is tagged.  Once tagged, travis will build and deploy, with the committed linked submodules.

The breaking change, is that the org.csstudio.product plugins, features, and pom.xml have been moved one directory deep, from the root directory to org.csstudio.product (ie was: org.csstudio.product now: org.csstudio.product/org.csstudio.product)

berryma4, 2017-12-20:
https://github.com/ControlSystemStudio/org.csstudio.product/pull/31

berryma4, 2017-12-20:
FYI: @shroffk @claudio-rosati @utzeln

claudio-rosati, 2017-12-20:
Ok. Thank you Eric. I’ll see what is the impact on my product when back from holidays in January. Merry Christmas

berryma4, 2017-12-22:
Removing deploy from travis to s3 from individual repos, and doing it from the org.csstudio.product repo.
To deploy, release to S3, you must:
- update the submodules
- run the org.csstudio.product/prepare-release.sh 
- tag with CSS-${version)
- push tag and branch for pull request
- merge pull request
- run org.csstudio.product/prepare-next-release.sh 

I see in tycho 1.1.0-SNAPSHOT there is support for the maven release plugin.  Once 1.1.0 is released, we can move to that.

closed 2018-01-08 by berryma4