ControlSystemStudio/cs-studio #2356: Beast enabling filter does not act until alarm configuration is opened

tashasummers, 2017-12-22:
In beast I have temporary alarms to alert me if a couple troublesome power supplies have a setpoint/readback mismatch. I put the power supply's ON status as an enabling filter. 

When I turn the power supplies off, I get all the mismatch alarms as active alarms until I open the alarm item configuration for the PV. Once the configuration window pops up the alarm disables and the circle goes gray. 

Once disabled, the alarm stays disabled when the power supply is turned on. When the configuration window is opened, the alarm re-enables.

kasemir, 2018-01-08:
Check if the alarm server is using
```
org.csstudio.platform.libs.epics/monitor=VALUE
```

Looks like in your case it's using 'ALARM', which means the alarm server only sees PV updates that change the alarm state. That's great to reduce network traffic for the alarm server, but effectively disables the filters.

berryma4, 2018-01-08:
This was the case.  Thank you!

closed 2018-01-08 by berryma4

kasemir, 2018-01-08:
The default is now "VALUE", plus nobody knows how to set the CA settings anyway after all the changes in the configuration handling, so you're safe going forward.