ControlSystemStudio/cs-studio #2359: BOY migration - led widgets save failing

shroffk, 2018-01-02:
```
shroffk@box64-2:/tmp$ 2018-01-02 09:41:16.696 WARNING [Thread 1] org.csstudio.opibuilder.editor.OPIEditor (processSaveFailedError) - File save failed
java.lang.NullPointerException
        at org.csstudio.opibuilder.widgets.editparts.LEDEditPart$11.propertyChange(LEDEditPart.java:166)
        at java.beans.PropertyChangeSupport.fire(PropertyChangeSupport.java:335)
        at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:327)
        at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:263)
        at org.csstudio.opibuilder.properties.AbstractWidgetProperty.firePropertyChange(AbstractWidgetProperty.java:89)
        at org.csstudio.opibuilder.properties.AbstractWidgetProperty.setPropertyValue_IgnoreOldValue(AbstractWidgetProperty.java:209)
        at org.csstudio.opibuilder.properties.RulesProperty.writeToXML(RulesProperty.java:175)
        at org.csstudio.opibuilder.persistence.XMLUtil.widgetToXMLElement(XMLUtil.java:96)
        at org.csstudio.opibuilder.persistence.XMLUtil.widgetToXMLElement(XMLUtil.java:104)
        at org.csstudio.opibuilder.persistence.XMLUtil.widgetToXMLString(XMLUtil.java:145)
        at org.csstudio.opibuilder.editor.OPIEditor.performSave(OPIEditor.java:967)
        at org.csstudio.opibuilder.editor.OPIEditor.doSave(OPIEditor.java:622)
        at org.eclipse.ui.internal.SaveableHelper$2.run(SaveableHelper.java:157)
        at org.eclipse.ui.internal.SaveableHelper$5.run(SaveableHelper.java:286)
        at org.eclipse.jface.operation.ModalContext.runInCurrentThread(ModalContext.java:437)
        at org.eclipse.jface.operation.ModalContext.run(ModalContext.java:353)
        at org.eclipse.ui.internal.WorkbenchWindow$14.run(WorkbenchWindow.java:2184)
        at org.eclipse.swt.custom.BusyIndicator.showWhile(BusyIndicator.java:70)
        at org.eclipse.ui.internal.WorkbenchWindow.run(WorkbenchWindow.java:2180)
        at org.eclipse.ui.internal.SaveableHelper.runProgressMonitorOperation(SaveableHelper.java:294)
        at org.eclipse.ui.internal.SaveableHelper.runProgressMonitorOperation(SaveableHelper.java:272)
        at org.eclipse.ui.internal.SaveableHelper.savePart(SaveableHelper.java:162)
        at org.eclipse.ui.internal.WorkbenchPage.saveSaveable(WorkbenchPage.java:3904)
        at org.eclipse.ui.internal.WorkbenchPage.saveEditor(WorkbenchPage.java:3918)
        at org.eclipse.ui.internal.handlers.SaveHandler.execute(SaveHandler.java:54)
        at org.eclipse.ui.internal.handlers.HandlerProxy.execute(HandlerProxy.java:295)
        at org.eclipse.ui.internal.handlers.E4HandlerProxy.execute(E4HandlerProxy.java:90)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.lang.reflect.Method.invoke(Method.java:497)
        at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:55)
        at org.eclipse.e4.core.internal.di.InjectorImpl.invokeUsingClass(InjectorImpl.java:282)
        at org.eclipse.e4.core.internal.di.InjectorImpl.invoke(InjectorImpl.java:264)
        at org.eclipse.e4.core.contexts.ContextInjectionFactory.invoke(ContextInjectionFactory.java:132)
        at org.eclipse.e4.core.commands.internal.HandlerServiceHandler.execute(HandlerServiceHandler.java:152)
        at org.eclipse.core.commands.Command.executeWithChecks(Command.java:494)
        at org.eclipse.core.commands.ParameterizedCommand.executeWithChecks(ParameterizedCommand.java:488)
        at org.eclipse.e4.core.commands.internal.HandlerServiceImpl.executeHandler(HandlerServiceImpl.java:210)
        at org.eclipse.e4.ui.bindings.keys.KeyBindingDispatcher.executeCommand(KeyBindingDispatcher.java:286)
        at org.eclipse.e4.ui.bindings.keys.KeyBindingDispatcher.press(KeyBindingDispatcher.java:507)
        at org.eclipse.e4.ui.bindings.keys.KeyBindingDispatcher.processKeyEvent(KeyBindingDispatcher.java:558)
        at org.eclipse.e4.ui.bindings.keys.KeyBindingDispatcher.filterKeySequenceBindings(KeyBindingDispatcher.java:378)
        at org.eclipse.e4.ui.bindings.keys.KeyBindingDispatcher.access$0(KeyBindingDispatcher.java:324)
        at org.eclipse.e4.ui.bindings.keys.KeyBindingDispatcher$KeyDownFilter.handleEvent(KeyBindingDispatcher.java:86)
        at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:84)
        at org.eclipse.swt.widgets.Display.filterEvent(Display.java:1605)
        at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1339)
        at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1366)
        at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1349)
        at org.eclipse.swt.widgets.Widget.sendKeyEvent(Widget.java:1378)
        at org.eclipse.swt.widgets.Widget.gtk_key_press_event(Widget.java:764)
        at org.eclipse.swt.widgets.Control.gtk_key_press_event(Control.java:3465)
        at org.eclipse.swt.widgets.Composite.gtk_key_press_event(Composite.java:801)
        at org.eclipse.swt.widgets.Widget.windowProc(Widget.java:2000)
        at org.eclipse.swt.widgets.Control.windowProc(Control.java:5827)
        at org.eclipse.swt.widgets.Display.windowProc(Display.java:5498)
        at org.eclipse.swt.internal.gtk.OS._gtk_main_do_event(Native Method)
        at org.eclipse.swt.internal.gtk.OS.gtk_main_do_event(OS.java:9545)
        at org.eclipse.swt.widgets.Display.eventProc(Display.java:1275)
        at org.eclipse.swt.internal.gtk.OS._g_main_context_iteration(Native Method)
        at org.eclipse.swt.internal.gtk.OS.g_main_context_iteration(OS.java:2495)
        at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:4149)
        at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$4.run(PartRenderingEngine.java:1121)
        at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
        at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1022)
        at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:150)
        at org.eclipse.ui.internal.Workbench$5.run(Workbench.java:693)
        at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
        at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:610)
        at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:148)
        at org.csstudio.utility.product.Workbench.runWorkbench(Workbench.java:103)
        at org.csstudio.startup.application.Application.startApplication(Application.java:265)
        at org.csstudio.startup.application.Application.start(Application.java:119)
        at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:196)
        at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
        at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
        at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:388)
        at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:243)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.lang.reflect.Method.invoke(Method.java:497)
        at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:673)
        at org.eclipse.equinox.launcher.Main.basicRun(Main.java:610)
        at org.eclipse.equinox.launcher.Main.run(Main.java:1519)
        at org.eclipse.equinox.launcher.Main.main(Main.java:1492)
```

shroffk, 2018-01-02:
While the rule can surely be replaced now with a better configured widget...the issue here is 

1. The file runs fine in runtime, the opi.runtime can properly understand the file however the editor fails to do so.

2. Upon repeated save attempts, every once in awhile, a save succeeds...

kasemir, 2018-01-08:
I can reproduce the error:
1) Add LED
2) Configure to have at least 3 states, so that there are state_color_0.. properties instead of the on/off colors
3) Add the rule similar to the screenshot that sets the state_color_0. Setting it to any color is OK, no need to use a named color.
4) Save the display, and you likely get the NPE.

Kunal is right that this type of rule should not be necessary.
The LED changes color based on the value of the primary PV. If you want to change between N colors, set the number of states to N. Don't add a rule to change colors, that violates the basic idea of the widget.

Still, there is a bug in the editor.

kasemir, 2018-01-08:
The NPE results from getting the old color, even though the old color isn't actually used..
Fixed it by checking for null.
So now initializeStateColor() is always called with oldColor==null, and then that color is ignored anyway, but I'm just fixing the bug, not BOY overall.

PR #2366 (2018-01-08): BOY: Avoid NPE in editor for LED
Fixes #2359
commit 7a454bef (2018-01-08): BOY: Avoid NPE in editor for LED

Fixes #2359
closed 2018-01-08 by berryma4