ControlSystemStudio/cs-studio #2361: Unable to build cs-studio

claudio-rosati, 2018-01-08:
Hello,

I hope you all started the new year well. After coming back from my Italian vacation I have the problem of not being able to build cs-studio.

I have 2 pipelines: the _css-ce_ one builds all the projects (`maven-osgi-bundles`, `cs-studio-thirdparty`, `diirt`, `cs-studio`, `org.csstudio.display.builder`, `org.csstudio.product`, and `org.csstudio.ess.product`) against the community master branch, and the _ess-css-development_ one using the ESS forked repositories.

Only when new features/fixes are merged into the community repositories **and** the _css-ce_ pipeline is successful, I synchronize the forked repositories with the community ones.

Now, the _css-cs_ pipeline fails building the `cs-studio` project:

```
...
00:16:58.679 [INFO] Compiling 78 source files to /home/jenkins/workspace/CS-Studio-CE/cs-studio/core/base/base-plugins/org.csstudio.vtype.pv/target/classes
00:16:59.650 [INFO] ------------------------------------------------------------------------
00:16:59.650 [INFO] Reactor Summary:
00:16:59.650 [INFO] 
00:16:59.650 [INFO] core .............................................. SUCCESS [0.567s]
00:16:59.650 [INFO] platform .......................................... SUCCESS [0.082s]
00:16:59.650 [INFO] platform-plugins .................................. SUCCESS [0.003s]
00:16:59.650 [INFO] org.csstudio.jre6.fragment ........................ SUCCESS [7.970s]
00:16:59.650 [INFO] utility ........................................... SUCCESS [0.005s]
00:16:59.650 [INFO] utility-plugins ................................... SUCCESS [0.004s]
00:16:59.650 [INFO] org.csstudio.data ................................. SUCCESS [1.359s]
00:16:59.650 [INFO] org.csstudio.data.test ............................ SUCCESS [8.268s]
00:16:59.650 [INFO] org.csstudio.java ................................. SUCCESS [0.363s]
00:16:59.651 [INFO] org.csstudio.logging .............................. SUCCESS [0.407s]
00:16:59.651 [INFO] org.csstudio.logging.test ......................... SUCCESS [6.414s]
00:16:59.651 [INFO] org.csstudio.logging.ui ........................... SUCCESS [0.600s]
00:16:59.651 [INFO] org.csstudio.openfile ............................. SUCCESS [0.191s]
00:16:59.651 [INFO] org.csstudio.rap.core ............................. SUCCESS [0.584s]
00:16:59.651 [INFO] diirt ............................................. SUCCESS [0.005s]
00:16:59.651 [INFO] diirt-plugins ..................................... SUCCESS [0.003s]
00:16:59.651 [INFO] org.csstudio.diirt.util.core.preferences .......... SUCCESS [0.279s]
00:16:59.651 [INFO] base .............................................. SUCCESS [0.002s]
00:16:59.651 [INFO] base-plugins ...................................... SUCCESS [0.001s]
00:16:59.651 [INFO] org.csstudio.vtype.pv ............................. FAILURE [0.856s]
00:16:59.651 [INFO] org.csstudio.startup .............................. SKIPPED
00:16:59.651 [INFO] org.csstudio.security ............................. SKIPPED
00:16:59.651 [INFO] org.csstudio.workspace ............................ SKIPPED
...
00:16:59.653 [INFO] ------------------------------------------------------------------------
00:16:59.653 [INFO] BUILD FAILURE
00:16:59.653 [INFO] ------------------------------------------------------------------------
00:16:59.653 [INFO] Total time: 5:40.137s
00:16:59.653 [INFO] Finished at: Mon Jan 08 11:07:56 CET 2018
00:17:02.227 [INFO] Final Memory: 470M/1572M
00:17:02.228 [INFO] ------------------------------------------------------------------------
00:17:02.228 [ERROR] Failed to execute goal org.eclipse.tycho:tycho-compiler-plugin:1.0.0:compile (default-compile) on project org.csstudio.vtype.pv: Compilation failure: Compilation failure:
00:17:02.228 [ERROR] /home/jenkins/workspace/CS-Studio-CE/cs-studio/core/base/base-plugins/org.csstudio.vtype.pv/src/org/csstudio/vtype/pv/jca/JCA_Preferences.java:[23]
00:17:02.228 [ERROR] import gov.aps.jca.jni.JNITargetArch;
00:17:02.228 [ERROR] ^^^^^^^^^^^^^^^
00:17:02.228 [ERROR] The import gov.aps.jca.jni cannot be resolved
00:17:02.228 [ERROR] /home/jenkins/workspace/CS-Studio-CE/cs-studio/core/base/base-plugins/org.csstudio.vtype.pv/src/org/csstudio/vtype/pv/jca/JCA_Preferences.java:[145]
00:17:02.228 [ERROR] final String jni_target = JNITargetArch.getTargetArch();
00:17:02.228 [ERROR] ^^^^^^^^^^^^^
00:17:02.228 [ERROR] JNITargetArch cannot be resolved
00:17:02.228 [ERROR] 2 problems (2 errors)
...
```

Any idea? Something dramatic changed since Dec. 15th?

kasemir, 2018-01-08:
I see the same error.
Was something mentioned at the recent CSS hangout regarding the build setup or the JCA libraries?

berryma4, 2018-01-08:
From the error, it looks like the new jca library is being used somehow.
Checking ...

berryma4, 2018-01-08:
The latest org.csstudio.product build seems to work:
https://travis-ci.org/ControlSystemStudio/org.csstudio.product/builds/325611627?utm_source=github_status&utm_medium=notification

There was this change to the org.csstudio.product repo (where the product directory was put in a separate directory in the repo):
https://github.com/ControlSystemStudio/org.csstudio.product
https://github.com/ControlSystemStudio/cs-studio/issues/2355

But, I don't understand why you would be getting the newer jca?  Could you check that this is the case, are you getting jca 2.4.1?

kasemir, 2018-01-08:
Starting a rebuild here.
So far I see that maven-osgi-bundles gets repository/target/repository/plugins/org.epics.jca_2.3.6.jar, i.e. the one that's been in use for a while.

kasemir, 2018-01-08:
Diirt creates `p2/target/repository/plugins/org.epics.jca_2.4.1.jar`, so if you build diirt, 3rd party, .. locally, that's how you get the updated JCA.

kasemir, 2018-01-08:
@shroffk is JCA 2.4.1 the latest and greatest, the one that combines JCA and CAJ? Then I'll update vtype.pv to use that.

shroffk, 2018-01-08:
Yes, it is the latest.
If you can wait for one more day...tomorrow I will have the new vtypes ready too.
Kunal

Sent via the Samsung GALAXY S®4, an AT&T 4G LTE smartphone

claudio-rosati, 2018-01-08:
I build maven-osgi-bundles, cs-studio-thirdparty, diirt in that order, and then cs-studio, display builder, ce-product and ess-product.

claudio-rosati, 2018-01-08:
I can wait few days...

berryma4, 2018-01-08:
@kasemir @claudio-rosati I see you are building diirt from master and not a released diirt.

berryma4, 2018-01-08:
After kunal does a diirt release, I was going to merge this branch:
https://github.com/ControlSystemStudio/cs-studio/tree/issue_2341
Although, it was made 20 days ago, and the libraries need to be all upped one (ie.jca 2.4.1)

kasemir, 2018-01-08:
Yes, I'm building all the 'master' branches.
.. which should be OK.

Submitted pull request that makes vtype.pv compile with JCA 2.4.1 (and would still work with 2.3.6).
For what it's worth, the JCALibrary reports old version:
```
public final class JCALibrary {
  static private final int VERSION=2;
  static private final int REVISION=3;
  // TODO version to be incremented - do not forget
  static private final int MODIFICATION=7;
```

I agree that updating pva is good, but no need to wait for that to remove the build error.

closed 2018-01-08 by berryma4

kasemir, 2018-01-08:
OK, build is again working for me when using all 'master' branches.

PR #2365 (2018-01-08): Jca 2361
Update vtype.pv to compile with JCA 2.4.

Fixes #2361